gazebosim/gazebo-classic #2718: Accept relative paths in URIs

chapulina, 2020-04-28:
The ability to pass relative paths in SDFormat's URIs was added on [this SDF PR](https://osrf-migration.github.io/sdformat-gh-pages/#!/osrf/sdformat/pull-requests/558) and implemented on Ignition Gazebo on [this PR](https://osrf-migration.github.io/ignition-gh-pages/#!/ignitionrobotics/ign-gazebo/pull-requests/433/page/1).

It would be nice if Gazebo 11, which uses SDF9, could also support relative paths, so more SDF files can be shared between Gazebo-classic and Ignition.

FracturedShader, 2020-05-18:
I was thinking that the easiest way would be by taking advantage of [AddFindFileCallback](https://github.com/osrf/gazebo/blob/d33e6429e3aa63181c3cef6d292370e6dc3f4320/gazebo/common/SystemPaths.hh#L126) in [InsertModelWidget](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/gui/InsertModelWidget.cc) where the call to [OnModelSelection](https://github.com/osrf/gazebo/blob/d33e6429e3aa63181c3cef6d292370e6dc3f4320/gazebo/gui/InsertModelWidget.cc#L295) would define the directory where to do additional searches (where the model.config lives) only for the duration of that model load.

chapulina, 2020-06-26:
@FracturedShader , since SDFormat 8 all elements carry the [path of the file that was used to load them](http://osrf-distributions.s3.amazonaws.com/sdformat/api/9.0.0/classsdf_1_1v9_1_1Element.html#abb2a719929e71bf157368d8fb379a4ae), so we can use that to resolve the relative URIs. See #2765 . This is only possible on Gazebo 11.

PR #2765 (2020-09-02): Accept relative paths in SDF files
Closes #2718 .

The main motivation for this is to allow more models to be reused between Gazebo-classic and Ignition to ease migration. Comments on this issue are relevant: https://github.com/ignitionrobotics/ign-fuel-tools/issues/77#issuecomment-648548920. I'm planning to document all this on a proper tutorial once the implementation is in.

This is a port of [this ign-gazebo PR](https://osrf-migration.github.io/ignition-gh-pages/#!/ignitionrobotics/ign-gazebo/pull-requests/433/page/1). 

---

## What's working

I locally modified the [Dumpster](https://app [... cut]
closed 2020-09-02 by chapulina