gazebosim/gazebo-classic #2720: Integrate ign-common's profiler

chapulina, 2020-04-29:
Add support for Ignition Common's [profiler](https://github.com/ignitionrobotics/ign-common/tree/master/profiler) to report time data from key sections of simulation execution.

We should measure and output profile data for at least:

* main thread
* physics
* rendering
* each sensor
* each plugin update

ahcorde, 2020-07-03:
A doubt that I have with this enhancement. It's that right now there is a diagnostic utility implemented in Gazebo

> Gazebo is currently instrumented with high-resolution diagnostic timers at several parts of the inner loop. The timer resolution depends on the hardware in use and is approximately 100ns on our test machines. To prevent performance degradation during normal usage, the timers are disabled unless the ENABLE_DIAGNOSTICS symbol is defined during compilation. For each simulation step, the elapsed time is measured and used to compute the following statistics incrementally: mean, minimum, maximum, and variance of each diagnostic timer. The statistics are computed using the math::SignalStats class. This will output the diagnostic timing data to the ~/.gazebo/diagnostics folder.

for example: 

https://github.com/osrf/gazebo/blob/6fd426b3949c4ca73fa126cde68f5cc4a59522eb/gazebo/physics/ode/ODEPhysics.cc#L361

Should I remove this? or both utilities should live together?

ahcorde, 2020-07-06:
Draft PR https://github.com/osrf/gazebo/pull/2776

chapulina, 2020-07-07:
> Should I remove this? or both utilities should live together?

I think we can leave the diagnostics stuff there now. The rationale is that all Gazebo versions have already been released and we don't want to change existing behaviour.

chapulina, 2020-10-08:
Done in #2776, #2813, #2836 and #2837

See tutorial: http://gazebosim.org/tutorials?tut=profiler

closed 2020-10-08 by chapulina