gazebosim/gazebo-classic #2726: re-release gazebo11 to focal using ogre-1.12

simonschmeisser, 2020-05-06:
Ubuntu Focal contains a much updated version 1.12.4 of OGRE which can be used by installing/depending on `libogre-1.12-dev` instead of `libogre-1.9-dev`. Updating the rosdeps was rejected/postponed because gazebo11 was already released and build against libogre-1.9. See here for the discussion https://github.com/ros/rosdistro/pull/24448 

Is there anything I can do to help with a new release? I would like to avoid being stuck on ogre-1.9 from 2013 due to compatibility reasons once ROS Noetic is released ...

traversaro, 2020-05-06:
I guess the first step is  to try to compile Gazebo 11 against `libogre-1.12-dev` . I am compiling and usiing Gazebo 11 on Windows against the ogre1.12 provided by vcpkg, but in the past there were problem on using Gazebo 11 with Ogre 1.12 from conda-forge, see https://github.com/osrf/gazebo/issues/2700 .

j-rivero, 2020-05-06:
Hello Simon:

> Is there anything I can do to help with a new release? I would like to avoid being stuck on ogre-1.9 from 2013 due to compatibility reasons once ROS Noetic is released ...

Thanks for all your efforts. I share your idea that would be really nice to update our ancient dependency on ogre-1.9. I'm afraid that the change is really complicated at this precise moment.

I've been in discussing with the rest of Gazebo/Ignition team and with the ROS developer in charge of Noetic and we have some considerations to execute the change at this moment:

 * We have used the [REP3](https://www.ros.org/reps/rep-0003.html#noetic-ninjemys-may-2020-may-2025) as the guide to know which base dependencies are we using in Noetic. The ogre version until today has been 1.9 (although there has been a foot note I think that the intention was not to change it 3 weeks before releasing). Different projects use that value to prepare releases for Noetic.
 * Although some patches have landed in Gazebo11 for ogre 1.12 we doubt that it can even compile out of the box. That is speaking only about building, changing a core dependency outside of our development and testing cycle, without knowing the potential effects at runtime (note that Silvio provided with an example of rendering problems) we think is not a good idea.
 * Packages have been released for Focal using Ogre 1.9. The potential impact on the API/ABI that changing Ogre version can have in Gazebo makes the change really complicated.

I'm open to help with other options (custom packages using ogre-1.12 in a different repo or with a different name) but the changes and testing that ogre 1.12 would require to go officially into Gazebo11 seems almost impossible to me. Sorry for not bringing good news this time.

traversaro, 2020-05-06:
>  note that Silvio provided with an example of rendering problems

The one I was referring to ( https://github.com/osrf/gazebo/issues/2700 ) is  probably solved (I guess)  but there are indeed other regression related to the current support for Ogre 1.12 in Gazebo 11, in particular https://github.com/osrf/gazebo/issues/2686 that is not trivial to fix.

simonschmeisser, 2020-05-07:
I'm not trying to push this idea further but just wanted to add that the ogre maintainer is very helpful and quick in responding. I could however not make much sense of those archived issues so I guess if someone where to ask him for help, a more detailed issue would be needed

j-rivero, 2020-05-08:
> . I could however not make much sense of those archived issues so I guess if someone where to ask him for help, a more detailed issue would be needed

Thanks Simon! I'm happy to see other people working in having upstream versions ready in Debian. Closing this, I hope we can find a solution/agreement in the rest of the issues that are open.

closed 2020-05-08 by j-rivero