gazebosim/gazebo-classic #2727: include pck-config file in gazebo11 instead of gazebo11-dev

simonschmeisser, 2020-05-07:
This would allow to reduce the exec-depend on gazebo_dev in gazebo_ros to a build_depend (or build_export_depend to be safe of regressions) here: https://github.com/ros-simulation/gazebo_ros_pkgs/blob/00a0064a1477667dfed75c3d22d91f14224db3a1/gazebo_ros/package.xml#L30

fixing https://github.com/ros-simulation/gazebo_ros_pkgs/issues/323 

As a consequence gazebo_ros would no longer transitively exec depend on `libogre-1.9-dev`, allowing people to build rviz plugins (eg MoveIt visualization) while having gazebo_ros installed.

I would love to prepare a PR but couldn't find the packaging unfortunately

j-rivero, 2020-05-07:
This PR proposes something that goes against the [specific rules of Debian packaging](https://www.debian.org/doc/debian-policy/ch-sharedlibs.html#id12) where any mechanism related to building should be installed by the `-dev` package.

We are not following all Debian rules in our packaging for packages.osrfoundation.org but this seems an important piece to me. Note that Gazebo11 is already present in Debian Sid as an official package and some people would for sure try to run ROS on top of this packaging. This change would break use cases for current and future Debian based platforms that rely on downstream packaging done in the right way.

My feeling is that we should find a different way of achieving the goal, which seems correct to me. Maybe modifying `gazebo_ros` somehow to use something different than pkg-config files from gazebo .deb packages or adding a fallback option if they pkg-config files are not available?

> I would love to prepare a PR but couldn't find the packaging unfortunately

The repository is here: https://github.com/ignition-release/gazebo11-release

simonschmeisser, 2020-05-07:
Oh, I wasn't aware that this is released to official Debian as well.

> My feeling is that we should find a different way of achieving the goal, which seems correct to me. Maybe modifying `gazebo_ros` somehow to use something different than pkg-config files from gazebo .deb packages or adding a fallback option if they pkg-config files are not available?

Under this condition I totally agree, would falling back to a hardcoded list of default locations if pkg-config is not found work?

simonschmeisser, 2020-05-07:
I opened a PR under https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1100

closed 2020-05-07 by simonschmeisser