gazebosim/gazebo-classic #2731: Missing gazebo9 9.13 release for arm64 (Ubuntu 18.04) in the osrf repository

petrlmat, 2020-05-11:
When installing `gazebo9` from the osrf repository on arm64 system running Ubuntu 18.04, the latest available version is 9.8. However the `gazebo9-common` dependency has version 9.13, which results in unmet dependencies error. 

Probably related to https://github.com/osrf/gazebo/issues/2647

### Steps to reproduce on arm64 with Ubuntu 18.04
1. `sudo apt install gazebo9-common`

output:
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  gazebo9-common
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 40,4 MB of archives.
After this operation, 78,4 MB of additional disk space will be used.
Get:1 http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic/main arm64 gazebo9-common all 9.13.0-1~bionic [40,4 MB]
Fetched 40,4 MB in 50s (810 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package gazebo9-common.
(Reading database ... 191386 files and directories currently installed.)
Preparing to unpack .../gazebo9-common_9.13.0-1~bionic_all.deb ...
Unpacking gazebo9-common (9.13.0-1~bionic) ...
Setting up gazebo9-common (9.13.0-1~bionic) ...
```

2. `sudo apt install gazebo9`

output:
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo9 : Depends: gazebo9-common (= 9.8.0-1~bionic) but 9.13.0-1~bionic is to be installed
           Recommends: gazebo9-plugin-base but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

```

### Repository packages
http://packages.osrfoundation.org/gazebo/ubuntu-stable/dists/bionic/main/binary-arm64/Packages

Package: gazebo9
Version: 9.8.0-1~bionic

Package: gazebo9-common
Version: 9.13.0-1~bionic

moriarty, 2020-06-02:
I believe this is related to #2744 see my comment https://github.com/osrf/gazebo/issues/2744#issuecomment-635765711 re: the `repository_uploader_ng` job in Jenkins uploading one of the other architectures `i386` `amd64` or `arm64` will delete the previous common `_all` debians that the other architectures require.

j-rivero, 2020-10-05:
arm64 should have been released for 9.15.0, check https://github.com/ignition-tooling/release-tools/issues/298. Please reopen if needed.

closed 2020-10-05 by j-rivero