gazebosim/gazebo-classic #2732: Is there a way to disable fixed joint reduction?

toliver, 2020-05-11:
When it converts a URDF to SDF, Gazebo reduces successive links that are united via fixed joints into a single link.
Is there a way to tell Gazebo to not do that (at least for a specific joint that we are interested in)?

The main reason is that this makes it impossible to declare a force torque sensor attached to a fixed joint, as it is suggested in the last section of [this tutorial](http://gazebosim.org/tutorials?tut=force_torque_sensor&cat=sensors).

When I try to do that, the sensor is removed together with the joint, and it does not appear in the final SDF file.

traversaro, 2020-05-11:
You can use the `preserveFixedJoint` tag, that unfortunately it is not documented, see https://github.com/osrf/gazebo_tutorials/issues/91 .

toliver, 2020-05-15:
Thanks for your quick answer @traversaro . That has worked well, although I've bumped into some unexpected beahviour.
It seems that gazebo silently removes any links that don't contain an `<inertial>` element, even if those links were necessary as an attachment point for subsequent links.

Not sure if I should file this as a bug, but I would expect some kind of warning at least.

As an example, the following URDF:

```
<robot name='test_robot'>
  <link name='link1'>
    <inertial>
      <origin xyz='0.0 0.0 0.0' rpy='0.0 0.0 0.0'/>
      <mass value='1.0'/>
      <inertia ixx='1.0' ixy='0.0' ixz='0.0'
               iyy='1.0' iyz='0.0' izz='1.0'/>
    </inertial>
  </link>
  <link name='link2'>
    <visual>
      <origin xyz='0.0 0.0 0.0' rpy='0.0 0.0 0.0'/>
      <geometry>
        <cylinder radius="0.01" length="0.01"/>
      </geometry>
    </visual>
    <inertial>
      <origin xyz='0.0 0.0 0.0' rpy='0.0 0.0 0.0'/>
      <mass value='1.0'/>
      <inertia ixx='1.0' ixy='0.0' ixz='0.0'
               iyy='1.0' iyz='0.0' izz='1.0'/>
    </inertial>
    <collision>
      <geometry>
        <box size="0.01 0.01 0.01"/>
      </geometry>
      <origin xyz='0.0 0.0 0.0' rpy='0.0 0.0 0.0'/>
    </collision>
  </link>
  <joint name='joint1_2' type='revolute'>
    <parent link='link1' />
    <child  link='link2' />
    <axis xyz="0 0 1" />
    <origin xyz='0.0 0.0 0.0' rpy='0.0 0.0 0.0' />
    <limit lower="0.0" upper="1.53" effort="-1" velocity="-1"/>
  </joint>
  
  <link name='camera_attachment_point'/>
  <joint name='joint1_camera_attachment_point' type='fixed'>
    <parent link='link1' />
    <child  link='camera_attachment_point' />
    <origin xyz='0.0 0.0 0.5' rpy='0.0 0.0 0.0' />
  </joint>

  <link name='camera'>
    <visual>
      <origin xyz='0.0 0.0 0.0' rpy='0.0 0.0 0.0'/>
      <geometry>
        <cylinder radius="0.01" length="0.01"/>
      </geometry>
    </visual>
    <inertial>
      <origin xyz='0.0 0.0 0.0' rpy='0.0 0.0 0.0'/>
      <mass value='1.0'/>
      <inertia ixx='1.0' ixy='0.0' ixz='0.0'
               iyy='1.0' iyz='0.0' izz='1.0'/>
    </inertial>
    <collision>
      <geometry>
        <box size="0.01 0.01 0.01"/>
      </geometry>
      <origin xyz='0.0 0.0 0.0' rpy='0.0 0.0 0.0'/>
    </collision>
  </link>
  <joint name='joint_camera' type='fixed'>
    <parent link='camera_attachment_point' />
    <child  link='camera' />
    <origin xyz='0.0 0.0 0.0' rpy='0.0 0.0 0.0' />
  </joint>

  <gazebo reference='joint1_camera_attachment_point'>
    <preserveFixedJoint>true</preserveFixedJoint>
  </gazebo>
  <gazebo reference='joint_camera'>
    <preserveFixedJoint>true</preserveFixedJoint>
  </gazebo>
</robot>
```
Is converted to the following SDF:
```
<sdf version='1.6'>
  <model name='test_robot'>
    <link name='link1'>
      <pose frame=''>0 0 0 0 -0 0</pose>
      <inertial>
        <pose frame=''>0 0 0 0 -0 0</pose>
        <mass>1</mass>
        <inertia>
          <ixx>1</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>1</iyy>
          <iyz>0</iyz>
          <izz>1</izz>
        </inertia>
      </inertial>
    </link>
    <link name='link2'>
      <pose frame=''>0 0 0 0 -0 0</pose>
      <inertial>
        <pose frame=''>0 0 0 0 -0 0</pose>
        <mass>1</mass>
        <inertia>
          <ixx>1</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>1</iyy>
          <iyz>0</iyz>
          <izz>1</izz>
        </inertia>
      </inertial>
      <collision name='link2_collision'>
        <pose frame=''>0 0 0 0 -0 0</pose>
        <geometry>
          <box>
            <size>0.01 0.01 0.01</size>
          </box>
        </geometry>
      </collision>
      <visual name='link2_visual'>
        <pose frame=''>0 0 0 0 -0 0</pose>
        <geometry>
          <cylinder>
            <length>0.01</length>
            <radius>0.01</radius>
          </cylinder>
        </geometry>
      </visual>
    </link>
    <joint name='joint1_2' type='revolute'>
      <child>link2</child>
      <parent>link1</parent>
      <axis>
        <xyz>0 0 1</xyz>
        <limit>
          <lower>0</lower>
          <upper>1.53</upper>
          <effort>-1</effort>
          <velocity>-1</velocity>
        </limit>
        <dynamics>
          <spring_reference>0</spring_reference>
          <spring_stiffness>0</spring_stiffness>
        </dynamics>
        <use_parent_model_frame>1</use_parent_model_frame>
      </axis>
    </joint>
  </model>
</sdf>
```

Where the `camera_attachment_point` has been removed and the subsequent `camera` link has also been removed.

traversaro, 2020-05-16:
Hi @toliver, I guess this is an interesting corner case in the URDF --> SDF conversions, and I guess indeed the correct way to proceed is probably to open a dedicated issue in https://github.com/osrf/sdformat . Related issues are: 
*  https://github.com/osrf/sdformat/issues/199
* https://github.com/osrf/sdformat/issues/105

traversaro, 2020-05-16:
In that specific case, a possible workaround is just to directly attach `camera` to `link2`.

toliver, 2020-05-18:
Thanks @traversaro . It seems that https://github.com/osrf/sdformat/issues/199 is the correct place to continue this discussion.
I'm closing this issue.

closed 2020-05-18 by toliver