gazebosim/gazebo-classic #2744: unable to install gazebo9 from packages.osrfoundation.org

moriarty, 2020-05-29:
@j-rivero & @chapulina originally I reported this directly to @jacobperron: I thought it was due to having both ROS & OSRF packages repositories setup but it it seems the problem is even simpler to reproduce.

```Dockerfile
FROM ubuntu:18.04

RUN apt-get update \
 && apt-get install -q -y --no-install-recommends \
    gnupg2 \
    lsb-release

RUN apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys D2486D2DD83DB69272AFE98867170598AF249743 \
 && . /etc/os-release \
 && echo "deb http://packages.osrfoundation.org/gazebo/$ID-stable `lsb_release -sc` main" > /etc/apt/sources.list.d/gazebo-latest.list

RUN apt-get update \
 && apt-get install -q -y --no-install-recommends \
  gazebo9
```

```
Step 4/4 : RUN apt-get update  && apt-get install -q -y --no-install-recommends   gazebo9
 ---> Running in 1af1e68e4727
Get:1 http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic InRelease [4256 B]
Hit:2 http://archive.ubuntu.com/ubuntu bionic InRelease
Get:3 http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic/main amd64 Packages [63.9 kB]
Get:4 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
Get:5 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]
Get:6 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]
Fetched 320 kB in 2s (146 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo9 : Depends: gazebo9-common (= 9.13.0-1~bionic) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
The command '/bin/sh -c apt-get update  && apt-get install -q -y --no-install-recommends   gazebo9' returned a non-zero code: 100
```

```
root@80d206d1d447:/# policy gazebo9
bash: policy: command not found
root@80d206d1d447:/# apt policy gazebo9
gazebo9:
  Installed: (none)
  Candidate: 9.13.0-1~bionic
  Version table:
     9.13.0-1~bionic 500
        500 http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic/main amd64 Packages
     9.0.0+dfsg5-3ubuntu1 500
        500 http://archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
root@80d206d1d447:/# apt policy gazebo9-common
gazebo9-common:
  Installed: (none)
  Candidate: 9.13.1-1~bionic
  Version table:
     9.13.1-1~bionic 500
        500 http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic/main amd64 Packages
     9.0.0+dfsg5-3ubuntu1 500
        500 http://archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
```

From here: http://packages.osrfoundation.org/gazebo/ubuntu-stable/pool/main/g/gazebo9/ I don't see any version of gazebo9 on bionic and amd64 that can be pinned to either as a temporary work around.

moriarty, 2020-05-29:
It looks like a Jenkins is already setup to catch this and it failed and notified a mailing list and @scpeters. 

https://build.osrfoundation.org/view/main/view/GazeboInstall/job/gazebo-install-gazebo9_pkg-bionic-amd64/127/console#console-section-5

```
The following packages have unmet dependencies:
 gazebo9 : Depends: gazebo9-common (= 9.13.0-1~bionic) but it is not going to be installed
 libgazebo9-dev : Depends: gazebo9-common (= 9.13.0-1~bionic) but it is not going to be installed
```

It's hard to follow the events which trigger the build jobs. Is it intended that when:
`gazebo9-common_9.13.1-1~bionic_all.deb` is uploaded that
`gazebo9-common_9.13.0-1~bionic_all.deb` is removed?

This job, based on it's Jenkins name and parameters
https://build.osrfoundation.org/view/debbuild/job/repository_uploader_ng/13851/
```
gazebo9 9.13.1-1(ubuntu/bionic::i386)
upload to: stable
RTOOLS_BRANCH: master
```
seems from it's name and it should only upload `i386` or `_all` debians. But instead it also deletes old ones including `_all`

```
Deleting files no longer referenced...
deleting and forgetting pool/main/g/gazebo9/gazebo9-common_9.13.0-1~bionic_all.deb
```

moriarty, 2020-05-29:
@j-rivero fixed it by re-triggering this build: https://build.osrfoundation.org/view/debbuild/job/gazebo9-debbuilder/545/

But I don't see any code or jenkins script fixes to prevent this from happening again so I will leave this issue open.

pauldinh, 2020-08-08:
I believe this problem has reared its head again. Attempting to install gazebo9 on 18.04 results in:

```
✘  15:40 ~ $ date
Sat Aug  8 15:41:00 CDT 2020
✔  15:41 ~ $ sudo apt install gazebo9
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo9 : Depends: gazebo9-common (= 9.13.2-1~bionic) but it is not going to be installed
           Recommends: gazebo9-plugin-base but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

And the relevant bits in http://packages.osrfoundation.org/gazebo/ubuntu-stable/pool/main/g/gazebo9/ show
```
gazebo9-common_9.14.0-1~bionic_all.deb             08-Aug-2020 06:21            40349480
gazebo9-dbg_9.13.2-1~bionic_amd64.deb              21-Jul-2020 01:45           194913748
gazebo9-doc_9.14.0-1~bionic_all.deb                08-Aug-2020 06:22                5100
gazebo9-plugin-base_9.13.2-1~bionic_amd64.deb      21-Jul-2020 01:46             1383700
gazebo9_9.13.2-1~bionic_amd64.deb                  21-Jul-2020 01:45             3898924
libgazebo9-dev_9.13.2-1~bionic_amd64.deb           21-Jul-2020 01:46              614088
libgazebo9_9.13.2-1~bionic_amd64.deb               21-Jul-2020 01:46             8131824
```

chapulina, 2020-08-08:
It looks like we had several issues while shipping the 9.14.0 release.

> I don't see any code or jenkins script fixes to prevent this from happening again

We're having issues like this too often, we'll bump the priority for a more permanent fix.

chapulina, 2020-08-08:
I just confirmed locally that Bionic and Focal are working.

Xenial is still running: [image]

dustingooding, 2020-08-09:
Confirmed on our Bionic+Melodic CI system that builds from a basic image, installs things from the PPA via rosdep, compiles code that depends on gazebo, and performs unit tests.  LGTM

moriarty, 2020-08-28:
@j-rivero & @chapulina thanks for looking into this!

AbelDengGang, 2020-10-04:
I have a simlar problem on 9.15. I just add source from  [http://gazebosim.org/tutorials?tut=install_ubuntu&cat=install](http://gazebosim.org/tutorials?tut=install_ubuntu&cat=install) 
Here's the gazebo9 version from this source:
`
gang@gang-Ubuntu:~$ sudo apt-cache madison gazebo9
   gazebo9 | 9.15.0-1~bionic | http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic/main amd64 Packages
   gazebo9 | 9.0.0+dfsg5-3ubuntu1+ppa2 | http://packages.ros.org/ros/ubuntu bionic/main amd64 Packages
   gazebo9 | 9.0.0+dfsg5-3ubuntu1 | http://cn.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
`

When I want to install it, it report error:
`
gang@gang-Ubuntu:~$ sudo apt install gazebo9 gazebo9-common
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo9 : Depends: gazebo9-common (= 9.15.0-1~bionic) but 9.15.0-2~bionic is to be installed
           Recommends: gazebo9-plugin-base
E: Unable to correct problems, you have held broken packages.
`
Does the server doesn't update intime?

gavanderhoorn, 2020-10-05:
This is a "me too", but a `+1` didn't seem right, so here is a comment instead.

I'm trying to build a Docker image based on `osrf/ros:kinetic-desktop` and then install Gazebo 9 in it from the OSRF repositories.

`apt` fails to be able to do that, with I believe the exact same error message as reported by @moriarty in https://github.com/osrf/gazebo/issues/2744#issuecomment-635765711, *except* now the version is `..2~xenial_all` vs `..1~xenial_all`.

The main `gazebo9` `.deb` appears to depend on `..1~xenial_all`, while only `..2~xenial_all` is available from the repository.

@j-rivero @chapulina: are some files missing or should the `control` file have been updated?

j-rivero, 2020-10-05:
release the update version on Bionic should fix the problem with this version: [image]

> @j-rivero @chapulina: are some files missing or should the `control` file have been updated?

Let's relax the dependency on the common package and we will update it if we need users to update it by some reason in the future. This should avoid the problem in the future: https://github.com/ignition-release/gazebo9-release/pull/9

gavanderhoorn, 2020-10-05:
Just making sure: I'm actually running into this on Xenial, not Bionic.

But Bionic could probably benefit from the change as well.

j-rivero, 2020-10-05:
> Just making sure: I'm actually running into this on Xenial, not Bionic.
> 
> But Bionic could probably benefit from the change as well.

Bionic and Xenial should be now fixed.The change is in place starting with `9.15.0-3` but the problem could be still present until all the previous versions are replaced. The issue https://github.com/ignition-tooling/release-tools/issues/307 should track the final solution to this problem. My apologies for the inconvenience.

gavanderhoorn, 2020-10-05:
Just ran a successful build.

Thanks @j-rivero for the quick turn-around :+1:

klaxalk, 2021-05-12:
Hey, I am experiencing this issue in Github Actions since May 7 on both Melodic and Noetic. Noetic got "fixed" by first adding the osfr apt sources for ignition. However, I am unable to work around that on Melodic. Any ideas?

https://github.com/ctu-mrs/uav_core/pull/21/checks?check_run_id=2546811621

traversaro, 2021-05-12:
> Hey, I am experiencing this issue in Github Actions since May 7 on both Melodic and Noetic. Noetic got "fixed" by first adding the osfr apt sources for ignition. However, I am unable to work around [...]

Possibly related issue (slightly different, but started from May 7 as well): https://github.com/osrf/gazebo/issues/3001 .

j-rivero, 2021-05-13:
> Possibly related issue (slightly different, but started from May 7 as well): #3001 .

I can not reproduce the problem, let's follow the investigation in #3001