gazebosim/gazebo-classic #2747: gazebo11 StaticMapPlugin not loading map tiles properly

Jaeyoung-Lim, 2020-06-03:
When using the `StaticMapPlugin`, the plugin is able to download the map tiles, but gazebo doesn't seem to disable it properly. 

[image]

- Usage followed instructions by [tuorial](http://gazebosim.org/tutorials?tut=static_map_plugin&cat=build_world)
- Related: https://answers.gazebosim.org/question/25083/gazebo-static-map-plugin-doesnt-work/?answer=25240#post-id-25240

Jaeyoung-Lim, 2020-06-29:
@chapulina Would you have any idea what is going on? How can we fix this?

chapulina, 2020-06-29:
I don't have much time to try to replicate this. Some ideas that may help:

* The console messages suggest that the tiles were correctly downloaded, do they look correct on their own? Are they valid images?
* Just making sure, no other messages are printed to the console? I'd expect some red errors.

BaltashovIlia, 2020-12-01:
Hi, @chapulina 

I have the same problem. The problem for me was that my Google maps API key stopped working.

At the same time, StaticMapPlugin did not give any errors or warnings. I think it's worth adding a key validation check, if possible, of course.

Jaeyoung-Lim, 2020-12-01:
@BaltashovIlia Did updating the keys solve the issue for you?

BaltashovIlia, 2020-12-01:
Yes

chapulina, 2020-12-01:
> I think it's worth adding a key validation check

Sounds like a good idea, [here](https://github.com/osrf/gazebo/blob/gazebo11/plugins/StaticMapPlugin.cc)'s the plugin code.

Jaeyoung-Lim, 2020-12-01:
@BaltashovIlia I don't think we have the same issue then. Otherwise, I wouldn't have been able to download the map tiles

BaltashovIlia, 2020-12-01:
@Jaeyoung-Lim 

Actually, if you have invalid Static Map API key then StaticMapPlugin will download tiles and even create png files. The problem is that you revived text with error message from google maps not image. Not the most obvious behavior )=.

To check it please opem .gazebo/models/map_satellite_*/textures/*.img. If you could not open the file, then there is a high probability it was in the api key.

To check your api key, put this https request in brouser: 

To check the api key, put this request with your key in the browser:

https://maps.googleapis.com/maps/api/staticmap?center=Albany,+NY&zoom=13&scale=1&size=600x300&maptype=roadmap&key=API_KEY&format=png&visual_refresh=true

kyuhyong1, 2024-06-14:
I applied API restrictions to allow Maps Static API and it is working.
Restrict key
1 API
Selected APIs:
Maps Static API
[image]