gazebosim/gazebo-classic #2750: gazebo plugin

qycqycqyc, 2020-06-08:
Could you tell me How to write gazebo plugin by myself,I want to learn it to write my own plugins.I had learnt gazebo tutorials,but there are not grammars.

Jaeyoung-Lim, 2020-06-08:
http://gazebosim.org/tutorials/?tut=plugins_hello_world

Jaeyoung-Lim, 2020-06-08:
@qycqycqyc Did you try following the tutorials and running the example plugin? For me this was enough to get started writing plugins for my own need

qycqycqyc, 2020-06-08:
@Jaeyoung-Lim Thanks,but I want to learn model plugin,which includes sdf,robotnamespace link etc.It is to hard if no grammars.

Jaeyoung-Lim, 2020-06-08:
@qycqycqyc I don't understand what you mean "grammars". There is no definition of what a grammar means in the gazebo API.

You can try looking at this plugin as an example: https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_gps_plugin.cpp

qycqycqyc, 2020-06-08:
@Jaeyoung-Lim Thank you very much,I see parameter 'robotnamespace' in your code, Could you tell me the function of it?

if (_sdf->HasElement("robotNamespace")) { namespace_ = _sdf->GetElement("robotNamespace")->Get<std::string>(); } else { gzerr << "[gazebo_gps_plugin] Please specify a robotNamespace.\n"; }

If not use robotnamespace in urdf,how to change it?

Jaeyoung-Lim, 2020-06-08:
@qycqycqyc We use it to parse the information from the sdf file and for mapping topics

closed 2020-06-21 by qycqycqyc