gazebosim/gazebo-classic #2752: Enable DART support in http://packages.osrfoundation.org's Gazebo11 binaries?

traversaro, 2020-06-08:
Hi @j-rivero, I had a small question on the Gazebo Ubuntu packages hosted on http://packages.osrfoundation.org, let me know if this is the proper place were to discuss this or if it is better to move it in https://github.com/ignition-release/gazebo11-release or somewhere else. 

There is a growing interested in using DART for iCub simulations on Gazebo11, especially for aspects related to manipulation and for providing a smooth upgrade path to Ignition Gazebo that will use DART as its default physic engine. While using DART  is relatively easy on macOS (where DART is an enabled dependency of gazebo11's homebrew bottle) or Windows (were we roll out our own binaries, so we can managed them easily), it involve a non-trivial complexity for the standard Ubuntu user, as at the moment it is necessary to compile Gazebo from source  to be able to use DART. 

As now the DART binary packages are available in http://packages.osrfoundation.org as they are a required (transitive) dependencies for `ignition-citadel`, I was wondering if it could make sense/be possible to enable DART support in Gazebo11 binaries  distributed from http://packages.osrfoundation.org . 
If that is possible at all, do you think there is something that as a community we can do to help on this? 

cc @Nicogene @xEnVrE @pattacini

j-rivero, 2020-06-09:
> Hi @j-rivero, I had a small question on the Gazebo Ubuntu packages hosted on http://packages.osrfoundation.org, let me know if this is the proper place were to discuss this or if it is better to move it in https://github.com/ignition-release/gazebo11-release or somewhere else.

Hi Silvio! I think the issue is fine here.

> As now the DART binary packages are available in http://packages.osrfoundation.org as they are a required (transitive) dependencies for `ignition-citadel`, I was wondering if it could make sense/be [...]

It has been always our intention to build Gazebo packages with all the support available so +1, I'll do some checks with the team and produce a prerelease to check that we are not going to break the ABI.

traversaro, 2020-07-05:
> It has been always our intention to build Gazebo packages with all the support available so +1, I'll do some checks with the team and produce a prerelease to check that we are not going to break the [...]

Great, thanks a lot (I thought I had  replied to this issue but apparently I did not). If there is anything we can do to help, please let us know, thanks!

j-rivero, 2020-07-07:
Thanks for the offer Silvio, I'm waiting for https://github.com/osrf/gazebo/pull/2770 and https://github.com/osrf/gazebo/pull/2734 to go in before doing the prerelease. Hopefully during this same week.