gazebosim/gazebo-classic #2756: What are the Magic numbers in the ActorPlugin?

mkhansenbot, 2020-06-13:
I'm trying to use the actor plugin in my gazebo world, and I'm able to follow the tutorial here: 
http://gazebosim.org/tutorials?tut=actor&cat=build_robot#Actors

I can put the actors in an empty world, no problem. Now I'm trying to put them in a occupied world and I can't figure out what to put for the pose and target that work? The only target value that seems to work is the one given in the example, which is also hardcoded in this line:
https://github.com/osrf/gazebo/blob/d9f480e07b9d3216ecda294691e3d5f86547951a/plugins/ActorPlugin.cc#L89

Also, I noticed that later in the code there are some other magic numbers being used in the random target generation code:
```
    newTarget.X(ignition::math::Rand::DblUniform(-3, 3.5));
    newTarget.Y(ignition::math::Rand::DblUniform(-10, 2));
```
Please help me understand what all these numbers mean? My best guess is that the random number generator is keeping the target area bounded between (-3,-10) and (3.5, 2). If so, why? Shouldn't those be the initial pose and target pose values?

mkhansenbot, 2020-06-13:
Oh and one other thing, I noticed that for some reason when I use the ActorPlugin, the person walking is slightly above the ground in the empty world.

chapulina, 2020-06-15:
The short answer is that there's a lot hardcoded specifically for those demos. As we migrate things to Ignition we're making sure to remove these magic numbers.

mkhansenbot, 2020-06-15:
Thanks for the response. I'm glad to hear these will be cleaned up when moving to Ignition. Do you have any timeline for that?

chapulina, 2020-06-15:
Basic actor support is already available on Ignition Citadel and control from plugins is already on `master`, which will become Ignition Dome in September. The implementation is spread across `ign-common`, `ign-rendering` and `ign-gazebo`.

Here's an example of an actor plugin that was ported to Ignition and shouldn't have any magic numbers that aren't configurable: https://github.com/ignitionrobotics/ign-gazebo/pull/157.

mkhansenbot, 2020-07-14:
Thanks, sorry for the delayed response, I'm satisfied with this answer and closing this

closed 2020-07-14 by mkhansenbot