gazebosim/gazebo-classic #2757: Boost bind global placeholders warning when compiling against Boost 1.73

traversaro, 2020-06-15:
When compiling Gazebo against Boost 1.73, a lot of compilation units generate this warnings:
~~~
C:\idjl-deps\vcpkg\installed\x64-windows\include\boost/bind.hpp(41): message : The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior. 
~~~

I do now know when this  header will be actually removed, but if it is ever remove from boost::bind that will eventually result in Gazebo compilation failure. 
See https://github.com/boostorg/bind/commit/5c21b6a5251523453ef3d02cc7e44dbfb31275eb .

scpeters, 2020-06-15:
We have partially migrated to using `std::bind` but didn't finish. I think we could continue that migration where it doesn't affect API / ABI

traversaro, 2020-08-10:
Related PR: https://github.com/osrf/gazebo/pull/2809 .

j-rivero, 2020-12-09:
I think this was fixed. Please reopen/comment if it was not.

closed 2020-12-09 by j-rivero