gazebosim/gazebo-classic #2766: Real-time performance of publishing marker messages

lum-supakorn, 2020-06-26:
I have been trying to find a way to visualize vectors (forces and moments on my model) so that debugging is easier. My current approach is to draw lines representing vectors using Gazebo markers and update them in every `OnUpdate()` loop of my model plugin. The major issue of this approach is that `node.Request("/marker", markerMsg)` is very sluggish and would soon freeze the whole program. Is there a better way to approach this visualization problem?

traversaro, 2020-08-28:
This may be a regression in Gazebo 11: we have a (unfortunately private) plugin that uses extensively the `/marker` topic, and it is working fine on Gazebo9/10, and instead crashes Gazebo 11 . Commenting the use of `/marker` solves the issue, at the cost of loosing the visualization @HosameldinMohamed  .

traversaro, 2021-06-30:
I was able to produce a MRE (minimum reproducible example) (actually two). The MREs can be found at https://gist.github.com/traversaro/bac589712c2e14e0722777c6b61dc9cc .

In that gist, there are two examples:
* `gazebo_2766_mre_small_leak.cc`  Example based on https://github.com/osrf/gazebo/tree/2e42eb91e7638da54b36dbb8fdd4ae5e3a310d6d/examples/stand_alone/marker, just modified to run in a loop instead of running once. This example result in a small leak in Gazebo11, that however may not be noticiable unless a user run a simulation for several minutes.
* `gazebo_2766_mre_big_leak.cc`  This example is based on the IIT's private plugin that I mentioned in https://github.com/osrf/gazebo/issues/2766#issuecomment-682378244 . This example result in a big leak in Gazebo11, that quickly makes the Gazebo simulator unstable and crashes it. I still need to check this example in Gazebo9, but based on the experience of the plugin, I would expect that even if a leak it is there, it is much smaller then the one present in Gazebo11, so the user may not notice it if they just run the simulation for a short amount of minutes.

To reproduce this results, do the following:
* Install Gazebo 11 with apt on Ubuntu 20.04 (I have  11.6.0)
* On one terminal, monitor the used memory with `watch -n 1 free -m`
* On another terminal, download, compile and run the mre:
~~~
git clone https://gist.github.com/traversaro/bac589712c2e14e0722777c6b61dc9cc gazebo_2766_mre
cd gazebo_2766_mre
mkdir build
cd build
cmake ..
make
# Check first the example with a small leak
./gazebo_2766_mre_small_leak 
# Then the big leak
./gazebo_2766_mre_big_leak 
~~~

traversaro, 2021-06-30:
An example of the mre running ony my system:

https://user-images.githubusercontent.com/1857049/123989881-98320180-d9c9-11eb-9a69-142719d4a0ad.mp4

traversaro, 2021-06-30:
For what regards the `MarkerManager` class, the only commit of difference between Gazebo9 and Gazebo11 is https://github.com/osrf/gazebo/commit/bf0823e2ddd2cfcd0c1209aa9bc1b6e467cf3da4#diff-c43d4dd3897612b7f621b6ee1b75ddd485f840dae5b5f8ef77f8be18fffe4ac5 . Among the lines changed by that commit, the one that got my attention is https://github.com/osrf/gazebo/commit/bf0823e2ddd2cfcd0c1209aa9bc1b6e467cf3da4#diff-c43d4dd3897612b7f621b6ee1b75ddd485f840dae5b5f8ef77f8be18fffe4ac5L214 . In particular, it seems that in Gazebo11 a new marker is always created if somebody does an operation with a marked with id 0. That could explain what we see, but I will check more on this tomorrow.

traversaro, 2021-07-01:
> 
> 
> For what regards the `MarkerManager` class, the only commit of difference between Gazebo9 and Gazebo11 is [bf0823e#diff-c43d4dd3897612b7f621b6ee1b75ddd485f840dae5b5f8ef77f8be18fffe4ac5](https://github.com/osrf/gazebo/commit/bf0823e2ddd2cfcd0c1209aa9bc1b6e467cf3da4#diff-c43d4dd3897612b7f621b6ee1b75ddd485f840dae5b5f8ef77f8be18fffe4ac5) . Among the lines changed by that commit, the one that got my attention is [bf0823e#diff-c43d4dd3897612b7f621b6ee1b75ddd485f840dae5b5f8ef77f8be18fffe4ac5L214](https://github.com/osrf/gazebo/commit/bf0823e2ddd2cfcd0c1209aa9bc1b6e467cf3da4#diff-c43d4dd3897612b7f621b6ee1b75ddd485f840dae5b5f8ef77f8be18fffe4ac5L214) . In particular, it seems that in Gazebo11 a new marker is always created if somebody does an operation with a marked with id 0. That could explain what we see, but I will check more on this tomorrow.

Indeed, this seems to be one problem. I tried to just avoid to use the id 0, that in Gazebo 9 was a normal ID, while on Gazebo 11 is synominous of "not initialized id", and so it is ignored and a new random ID is created. This was done modifying the line in `gazebo_2766_mre_big_leak` :
~~~
m_markerMsg.set_id(jet);
~~~
to 
~~~
m_markerMsg.set_id(jet+1);
~~~
(note that `jet` is an index that goes from 0 to 3, see https://gist.github.com/traversaro/bac589712c2e14e0722777c6b61dc9cc/revisions#diff-fbfd001186e77f112139ceb791cfafd3f0c61773781f8b8312e4186827b80d01R47). And doing that, indeed the big leak disappeared.

traversaro, 2021-07-01:
So, to reply to the original question by @lum-supakorn : 

>  The major issue of this approach is that node.Request("/marker", markerMsg) is very sluggish and would soon freeze the whole program. Is there a better way to approach this visualization problem?

Yes, avoid to use the marked id `0` as its semantics changed between Gazebo 9 and Gazebo 11. This should avoid to have a leak in the number of marker created, as in Gazebo 11 id `0` means "silently ignore the id, and create a new marked with a random ID".

traversaro, 2021-07-01:
By the way, it seems that this issue was detected by @rfn123 in https://github.com/osrf/gazebo/issues/2920 , but I was not aware of that issue.