gazebosim/gazebo-classic #2794: How to use multiple CPU cores

liangpu597, 2020-07-22:
I roslaunch a world and the it's very slow.
I check the cpu's use
[image]

and found only one core is run.
How to use multiple CPU cores? 
thank you!

LebinLiang, 2022-11-29:
I met the same problem...Did u fix it?

traversaro, 2022-12-01:
The behaviour of Gazebo Classic w.r.t. to CPU usage is highly depending on the world/models that are simulated and the simulation settings. Gazebo Classic is already using multiple threads for several concerns (physics engine, rendering, etc etc), but apparently in your case one of the threads is taking 100%. Understanding which thread is taking 100% is a world/model-specific question.

In general, transforming a part of a program from being single thread to use multiple thread (to exploit multiple CPUs) is a non-trivial transformation, so there is no single settng or option that will transform a part of a program to be exploit multiple CPUs. As Gazebo Classic is mantainance mode, and all future development is happening on the new Gazebo Sim ( https://gazebosim.org/home ), I do not think they will be future development on this area.

Anyhow, if you have speed problems there may be other ways to improve them. First of all, you may need to understand what is the bottleneck of your computation that is reaching 100% CPU usage. For doing that, I suggest to launch gzserver and gzclient separatly and check which one is using more CPU, to understand if the problem is in the physics engine (gzserver) or rendering (gzclient). If you could share your world, that would also help the debug. Anyhow, for this kind of question-like problems, I suggest you to open a  question on Gazebo Answers (https://answers.gazebosim.org/questions/) or on Robotics StackExchange (https://robotics.meta.stackexchange.com/), as soon Gazebo Answers will be migrated there (https://discourse.ros.org/t/planned-migration-of-ros-answers-and-gazebo-answers-to-robotics-stack-exchange/28068).

Let me know if we can close the issue given this info, thanks!