gazebosim/gazebo-classic #2805: Gazebo should print diagnostic messages on terminal

ydirson, 2020-07-29:
Sample situation: I have a gazebo instance already running, just happen to forget about it.  When I attempt to run another one, I'm left with absolutely no error message beside the `exit(255)` (yes, `exit(-1)` does not exist in real life):

```
$ gazebo -u
$ echo $?
255
$
```

The user has to dig through all logs in ~/.gazebo to find a `gzserver.log` that says `There is probably another Gazebo process running`.  It would be much more friendly to have that (and surely other server error messages) printed to stderr by default, at least when running through the `gazebo` wrapper.

chapulina, 2020-07-31:
Maybe you're looking for the `--verbose` argument?

chapulina, 2020-07-31:
I'm closing this, please reopen if you identify a bug on Gazebo.

closed 2020-07-31 by chapulina

ydirson, 2020-08-01:
Well, I must say i'm surprised I have to use `--verbose` to get errors.  Years of using Linux brought the expectation to get told about errors without the need to use a flag, and with that of `--verbose` giving additional informative details about what's happening.

ydirson, 2020-08-01:
Adjusted the issue title, but can't find the "reopen" button for some reason.

chapulina, 2020-08-03:
> i'm surprised I have to use --verbose to get errors

See https://github.com/osrf/gazebo/issues/1705. On Ignition, we support different logging levels and default to showing errors. On Gazebo this is unlikely to change.