gazebosim/gazebo-classic #2806: After loading world saved from non-initial state, "run" throws models through unexpected trajectories

ydirson, 2020-07-29:
I load this with `gazebo -u test-world.sdf` in 11.0.0, then hit the "run" button without resetting the positions first, the objects get thrown away by some unspecified force.

If OTOH I use Ctrl-R first, then I get the same behaviour I had before saving.

[test-world.sdf.gz](https://github.com/osrf/gazebo/files/4994956/test-world.sdf.gz)

chapulina, 2020-07-31:
> the objects get thrown away by some unspecified force.

It's not unspecified, there are velocities and accelerations specified under the `<state>` tag in your world. Remove that tag if you want everything to start from a rested state.

I recommend you take these questions first to https://answers.gazebosim.org/ if you're not sure they're bugs on Gazebo. That's the correct place to get technical help.

closed 2020-07-31 by chapulina

ydirson, 2020-08-01:
OK, though it surprises me that saving a world where everything has stopped moving under the sole effect of gravity would get saved with velocities and accelerations throwing them up in the air.