gazebosim/gazebo-classic #2807: Submodel include inconsistency ?

ydirson, 2020-07-29:
Working on understanding behaviour of the `<include>` feature, I modified the model from the nested_model tutorial, and changed it to use inclusion.  The changes from the tuto are designed to show the joint in action under the simple effect of gravity.

I expected that moving the submodel description into a separate model would yield the same result as a scene, but something I can't identify seems to influence the pose of the submodel.

Maybe this can be used as a basis to further extend the nested_model tuto, by illustrating the _Note on the include SDF element_.

[nested-include.tar.gz](https://github.com/osrf/gazebo/files/4995215/nested-include.tar.gz)

chapulina, 2020-07-31:
> but something I can't identify seems to influence the pose of the submodel.

Could you describe better what you're expecting to observe? They seem to do the same thing to me, I'm on Gazebo 11:

[image]

ydirson, 2020-08-01:
Here is what I get

chapulina, 2020-08-03:
What's the exact version you're using? See with `gazebo --version`. Also, the difference in lighting is suspicious. Are you getting any console errors with `--verbose`?

ydirson, 2020-08-03:
```
$ gazebo --version
Gazebo multi-robot simulator, version 11.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

$ apt-cache policy gazebo
gazebo:
  Installed: 11.0.0+dfsg1-4+b3
  Candidate: 11.0.0+dfsg1-4+b3
  Version table:
 *** 11.0.0+dfsg1-4+b3 990
        990 http://ftp.fr.debian.org/debian testing/main amd64 Packages
        500 http://ftp.fr.debian.org/debian unstable/main amd64 Packages
        100 /var/lib/dpkg/status
```

Hm, I do have quite a couple of them, in fact.

```
[Err] [InsertModelWidget.cc:417] Missing model.config for model "/home/yann/perso/robotics/gazebo_tutorials"
[Err] [InsertModelWidget.cc:417] Missing model.config for model "/home/yann/perso/robotics/sloth"
[Err] [InsertModelWidget.cc:417] Missing model.config for model "/home/yann/perso/robotics/test"
[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
[Err] [SdfFrameSemantics.cc:82] Error Code 16 Msg: Nested models are not yet supported by DOM objects, skipping model [model_00].
[Err] [SdfFrameSemantics.cc:82] Error Code 18 Msg: Child link with name[model_01::link_01] specified by joint with name[joint_00] not found in model with name[model_00].
[Err] [SdfFrameSemantics.cc:82] Error Code 23 Msg: FrameAttachedToGraph error, Non-LINK vertex with name [joint_00] is disconnected; it should have 1 outgoing edge in MODEL attached_to graph.
[Err] [SdfFrameSemantics.cc:82] Error Code 23 Msg: Graph has __model__ scope but sink vertex named [joint_00] does not have FrameType LINK when starting from vertex with name [joint_00].
[Err] [SdfFrameSemantics.cc:82] Error Code 18 Msg: Child link with name[model_01::link_01] specified by joint with name[joint_00] not found in model with name[model_00].
[Err] [SdfFrameSemantics.cc:82] Error Code 26 Msg: PoseRelativeToGraph error, Non-MODEL vertex with name [joint_00] is disconnected; it should have 1 incoming edge in MODEL relative_to graph.
[Err] [SdfFrameSemantics.cc:82] Error Code 26 Msg: PoseRelativeToGraph frame with name [joint_00] is disconnected; its source vertex has name [joint_00], but its source name should be __model__.
[Err] [Model.cc:99] Error Code 16 Msg: Nested models are not yet supported by DOM objects, skipping model [model_00].
[Err] [Model.cc:99] Error Code 18 Msg: Child link with name[model_01::link_01] specified by joint with name[joint_00] not found in model with name[model_00].
[Err] [Model.cc:99] Error Code 23 Msg: FrameAttachedToGraph error, Non-LINK vertex with name [joint_00] is disconnected; it should have 1 outgoing edge in MODEL attached_to graph.
[Err] [Model.cc:99] Error Code 23 Msg: Graph has __model__ scope but sink vertex named [joint_00] does not have FrameType LINK when starting from vertex with name [joint_00].
[Err] [Model.cc:99] Error Code 18 Msg: Child link with name[model_01::link_01] specified by joint with name[joint_00] not found in model with name[model_00].
[Err] [Model.cc:99] Error Code 26 Msg: PoseRelativeToGraph error, Non-MODEL vertex with name [joint_00] is disconnected; it should have 1 incoming edge in MODEL relative_to graph.
[Err] [Model.cc:99] Error Code 26 Msg: PoseRelativeToGraph frame with name [joint_00] is disconnected; its source vertex has name [joint_00], but its source name should be __model__.
```

Also note that in some cases, the joint seems to be completely inoperative, like in this world, made of several of those insert operations

azeey, 2020-08-05:
I was able to reproduce this on Gazebo11 and Gazebo9. I first added `nested-include` from the GUI. Then I added `nested` and that's when I saw the `nested-include` model unexpectedly.

The issue seems to be related how the joint in the `nested-include` model interacts with the spawning of the `nested` model. In my run, `nested-include` was named `model_00` and `nested` was named `model_00_0`. Inside `model_00`, the joint's child link is `model_00::model_01::link_01`. But inside `model_00_0`, the joint has the same child link, `model_00::model_01::link_01` when it should be `model_00_0::model_01::link_01`.

ydirson, 2020-08-09:
Maybe there is a link with this other behaviour ? https://answers.gazebosim.org/question/25620/action-at-a-distance-when-inserting-a-model-in-a-running-scene/