gazebosim/gazebo-classic #2810: Broken libgazebo7-dev package given last update to 7.16.1

mateus-amarante, 2020-08-03:
I think PR #2800 introduced a problem in `libgazebo7-dev` ubuntu package. When I run `sudo apt install libgazebo7-dev`

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgazebo7-dev : Depends: gazebo7-common (= 7.16.0-1~xenial) but it is not going to be installed
                  Depends: gazebo7-plugin-base (= 7.16.0-1~xenial)
E: Unable to correct problems, you have held broken packages.
```

It cannot resolve gazebo7-common (= 7.16.0-1~xenial), which is already in 7.16.1-1~xenial version

j-rivero, 2020-08-03:
There was a problem with the building infrastructure: https://build.osrfoundation.org/job/gazebo7-debbuilder/496. I've re triggered the build in https://build.osrfoundation.org/job/gazebo7-debbuilder/501/. Building now, should be fixed after it finishes successfully.

mateus-amarante, 2020-08-03:
Is it suppose to work now? I still get the same error

j-rivero, 2020-08-03:
> Is it suppose to work now? I still get the same error

It was supposed to ... not the best week for the infrastructure. I'm pretty sure it is working now.

mateus-amarante, 2020-08-03:
It is working now! Thank you very much!

closed 2020-08-03 by mateus-amarante