gazebosim/gazebo-classic #2820: Installing gazebo 11 from source on Ubuntu focal 20.04 - unable to locate libboost-signals-dev

ankitbhatia, 2020-08-11:
I'm trying to install gazebo 11 with dart support on ubuntu 20.04. 

I am attempting to fetch all dependencies as shown in the tutorial with 

```
wget https://raw.githubusercontent.com/ignition-tooling/release-tools/master/jenkins-scripts/lib/dependencies_archive.sh -O /tmp/dependencies.sh
GAZEBO_MAJOR_VERSION=11 ROS_DISTRO=noetic . /tmp/dependencies.sh
echo $BASE_DEPENDENCIES $GAZEBO_BASE_DEPENDENCIES | tr -d '\\' | xargs sudo apt-get -y install
```
But there is an error saying

E: Unable to locate package libboost-signals-dev

It seems that's not a package available in ubuntu 20.04 focal. Is there any way this could be resolved?

chapulina, 2020-08-11:
>  install gazebo 11 with dart support 

Please look at this issue for that: https://github.com/osrf/gazebo/issues/2752

j-rivero, 2020-08-13:
@ankitbhatia could you please check if using https://raw.githubusercontent.com/ignition-tooling/release-tools/30a12086e235c92c93a3b23150ea94a5e5fba1eb/jenkins-scripts/lib/dependencies_archive.sh solves the problem for you?

ankitbhatia, 2020-08-13:
@j-rivero I ended up using the binary release for gazebo and everything works now with dart support.

closed 2020-08-13 by ankitbhatia