gazebosim/gazebo-classic #2830: Blender/Collada imports look Grey'ish

HighPriest, 2020-08-23:
Collada .dae file with baked in material renders like this:
[image]
With grey ambient

skohlbr, 2020-08-24:
I encountered the same issue. The Blender export (using Blender > 2.80) looks like this:

>             <emission>
>               <color sid="emission">0.01 0.01 0.01 1</color>
>             </emission>
>             <diffuse>
>               <color sid="diffuse">0.64 0.6071135 0.007775764 1</color>
>             </diffuse>

What's missing is the `<ambient>` tag, which Gazebo apparently assumes to be some default value when it's missing. You can thus fix the issue by adding the tag manually in your Collada files (set the numeric value to some low/dark greyish value that works for you):

>             <ambient>
>               <color sid="ambient">0.05 0.05 0.05 1</color>
>             </ambient>

conradstevens, 2024-07-10:
Its worth a try defining the material through a `model.material` file.

closed 2024-12-30 by HighPriest