gazebosim/gazebo-classic #2841: Modifying Lens Flare effects to account for number of elements and shape

kjeppesen1, 2020-09-07:
There is interest in modifying the [LensFlareSensorPlugin](https://github.com/osrf/gazebo/blob/gazebo11/plugins/LensFlareSensorPlugin.cc) to allow the user to specify the number of lens elements, as well as the shape of the aperture, to produce a more realistic lens flare. From working backwards through the code, I _think_ that the flare is defined in [camera_lens_flare_fs](https://github.com/osrf/gazebo/blob/gazebo11/media/materials/programs/camera_lens_flare_fs.glsl) by user @iche033 . Which aspects of this code are responsible for controlling the shape and number of flares, and what is the methodology?

iche033, 2020-09-08:
yes lens flare is implemented in `camera_lens_flare_fs.glsl`. It is mostly an post processing effect and not physically based. So unfortunately it can't be configured based on properties of a real lens.

kjeppesen1, 2020-09-08:
@iche033 but it is possible to modify the post-processed effect given some input from a new plugin attribute? E.g. the user may specify 6 lens elements and a pentagonal aperture shape, and those values would be used by `camera_lens_flare_fs.glsl` to create a similar effect to the existing one, but with 6 five-sided artifacts.

iche033, 2020-09-08:
as an example, you can take a look at how the existing lens flare plugin passes the `scale` param to the shaders:
* setting scale from plugin: https://github.com/osrf/gazebo/blob/gazebo11/plugins/LensFlareSensorPlugin.cc#L113
* setting shader uniform param: https://github.com/osrf/gazebo/blob/gazebo11/gazebo/rendering/LensFlare.cc#L132
* make sure to define the param in material script: https://github.com/osrf/gazebo/blob/gazebo11/media/materials/scripts/gazebo.material#L1552

If you plan to change existing appearance / behavior of the current lens flare effect, I would recommend doing it in a separate plugin to not break existing code