gazebosim/gazebo-classic #2847: Changelog for 11.1.0 is missing `--lockstep` PRs

scpeters, 2020-09-23:
The 11.1.0 release included the `--lockstep` command-line parameter to enable synchronization of physics and sensor time stepping, but it's not included in the changelog. I've added it on a branch in https://github.com/scpeters/gazebo/commit/f34f1016c895df4261733d27b1a09d1b67aa39e1, but I didn't want to run all CI, so I created an issue for visibility and will cherry-pick directly to `gazebo11` if other maintainers approve.

mabelzhang, 2020-09-23:
Looks good to me.

Does the changelog for 9.14.0 also need to be updated?

scpeters, 2020-09-23:
> Looks good to me.
> 
> Does the changelog for 9.14.0 also need to be updated?

maybe; let's do that on the `gazebo9` branch and merge forward

scpeters, 2020-09-23:
ok, I'll go ahead and cherry-pick https://github.com/scpeters/gazebo/commit/f34f1016c895df4261733d27b1a09d1b67aa39e1 to `gazebo11`

scpeters, 2020-09-23:
> ok, I'll go ahead and cherry-pick [scpeters@f34f101](https://github.com/scpeters/gazebo/commit/f34f1016c895df4261733d27b1a09d1b67aa39e1) to `gazebo11`

done in https://github.com/osrf/gazebo/commit/458bbee3d8a3f5b4cb413d4ad549fb5891979790

scpeters, 2020-09-23:
> Looks good to me.
> 
> Does the changelog for 9.14.0 also need to be updated?

actually, it looks like it's already there, just needs to be merged forward

closed 2020-09-23 by scpeters