gazebosim/gazebo-classic #2855: unable to install gazebo9_9.15 on ubuntu 18 ,64bit PC

AbelDengGang, 2020-10-04:
I update source from [http://gazebosim.org/tutorials?tut=install_ubuntu&cat=install](http://gazebosim.org/tutorials?tut=install_ubuntu&cat=install), but it report error when I try to install it:
`gang@gang-Ubuntu:~$ sudo apt install gazebo9 gazebo9-common
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo9 : Depends: gazebo9-common (= 9.15.0-1~bionic) but 9.15.0-2~bionic is to be installed
           Recommends: gazebo9-plugin-base
E: Unable to correct problems, you have held broken packages.
`

From the [http://packages.osrfoundation.org/gazebo/ubuntu-stable/pool/main/g/gazebo9/](http://packages.osrfoundation.org/gazebo/ubuntu-stable/pool/main/g/gazebo9/) , I can find gazebo9_9.15.0-2~bionic_**arm64**.deb  and gazebo9-common_9.15.0-2~bionic_all.deb . Is gazebo9_9.15.0-2~bionic_**amd64**.deb missed for ubuntu 18?

deadln, 2020-10-04:
I have the same problem:

$ sudo apt install gazebo9
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo9 : Depends: gazebo9-common (= 9.15.0-1~bionic) but 9.15.0-2~bionic is to be installed
           Recommends: gazebo9-plugin-base
E: Unable to correct problems, you have held broken packages.

 I succeed to install gazebo9-common(9.15.0-2) manually, but it looks like that there is 9.15.0-1 version is supposed to be. I tried this:

$ sudo apt install gazebo9-common=9.15.0-1~bionic
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Version '9.15.0-1~bionic' for 'gazebo9-common' was not found

And there is no clue how to install this older version in entire Internet. And the same problem with gazebo9-plugin-base, but it depend on gazebo9:

$ sudo apt install gazebo9-plugin-base
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo9-plugin-base : Depends: gazebo9 (= 9.15.0-1~bionic)
E: Unable to correct problems, you have held broken packages.

GitSRealpe, 2020-10-05:
same here, just a week ago it was installing without problems with the same apt command
`sudo apt-get install ros-kinetic-gazebo9-*Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'ros-kinetic-gazebo9-dev' for glob 'ros-kinetic-gazebo9-*'
Note, selecting 'ros-kinetic-gazebo9-msgs' for glob 'ros-kinetic-gazebo9-*'
Note, selecting 'ros-kinetic-gazebo9-ros' for glob 'ros-kinetic-gazebo9-*'
Note, selecting 'ros-kinetic-gazebo9-plugins' for glob 'ros-kinetic-gazebo9-*'
Note, selecting 'ros-kinetic-gazebo9-ros-control' for glob 'ros-kinetic-gazebo9-*'
Note, selecting 'ros-kinetic-gazebo9-ros-pkgs' for glob 'ros-kinetic-gazebo9-*'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-kinetic-gazebo9-dev : Depends: libgazebo9-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.`

traversaro, 2020-10-05:
Example of this failure in CI: https://github.com/robotology/gazebo-yarp-plugins/runs/1207011683 .

v-lopez, 2020-10-05:
Same. 

It looks like gazebo9 version `9.15.0-2~bionic`is missing from here, only `9.15.0-1~bionic` is available http://packages.osrfoundation.org/gazebo/ubuntu-stable/pool/main/g/gazebo9/

Happening since last Friday.

gavanderhoorn, 2020-10-05:
I believe I've ran into the same problem, but on `16.04` (#2744).

Same version, just for `~xenial`.

j-rivero, 2020-10-05:
My bad sorry. release the update version on Bionic should fix the problem with this version:  [image]

Let's relax the dependency on the common package and we will update it if we need users to update it by some reason in the future. This should avoid the problem in the future: ignition-release/gazebo9-release#9

j-rivero, 2020-10-05:
Testing installation job for Xenial/amd64 [image]. Please reopen if needed.

closed 2020-10-05 by j-rivero

gavanderhoorn, 2020-10-05:
@j-rivero: I'm not up-to-date on the syncing policy for Gazebo. Will these builds be available immediately? Or after a specific sync?

j-rivero, 2020-10-05:
> @j-rivero: I'm not up-to-date on the syncing policy for Gazebo. Will these builds be available immediately? Or after a specific sync?

When releasing directly to stable, they are, yes. There are no sync procedures in Gazebo repositories (yet).

> This links to a Bionic build (ie: `gazebo-install-gazebo9_pkg-bionic-amd64`)?

sorry, [image]