gazebosim/gazebo-classic #2857: Do you have a plan to release gazebo11 for ubuntu focal with armhf in the osrf apt repository?

sktometometo, 2020-10-12:
I am using a ROS package which depends on the gazebo_ros package on armhf architecture.
Its test for ubuntu focal fails because gazebo11 package is not released for ubuntu focal with armhf. http://build.ros.org/job/Nbin_ufhf_uFhf__gazebo_ros__ubuntu_focal_armhf__binary/

Do you have a plan to release gazebo11 for ubuntu focal with armhf architecture?
Or is there anything for me to help it?

chapulina, 2020-10-13:
This issue is tracking the latest Gazebo 11.2.0 release: https://github.com/ignition-tooling/release-tools/issues/302

We haven't been paying too much attention to armhf because there aren't enough users to justify the effort needed. If you're willing to help, we welcome pull requests fixing the build.

I believe this file would also need to be removed: https://github.com/ignition-release/gazebo11-release/blob/master/focal/.releasepy_NO_ARCH_ARMHF

j-rivero, 2020-10-13:
Thanks for the message @sktometometo . Out of curiosity, which platform/board are you using?