gazebosim/gazebo-classic #2861: Incorrect sensor update rate in lockstep mode

iche033, 2020-10-21:
When two physics based sensors of same type but with different update rates are present in the world, both sensors end up with the same (higher) frame rate.

To reproduce:

1. create two sensors of different update rate, e.g. copy this [hokuyo model from gazebo_models](https://github.com/osrf/gazebo_models/tree/master/hokuyo) to a new model, named `hokuyo_100`, and change its update rate to 100. and place them into your gazebo models path, e.g. `~/.gazebo/models`

1. Launch gazebo and insert the original hokuyo model into the world. Check its update rate by using the Topic Visualization tool (Ctrl +T) and selecting the hokuyo topic under `LaserScanStamped` msg. The update rate should be 30Hz.

1. Insert the  `hokuyo_100` model into the world, and see that the update rate for the original hokuyo sensor jumps to 100Hz

peci1, 2023-10-12:
Fixed by #2863 in Gazebo 9, so can be closed? However, not fixed in 11!

peci1, 2024-11-16:
Now it's also fixed in 11.15.0 .

traversaro, 2024-11-16:
Great, thanks for bringing that up.

closed 2024-11-16 by traversaro