gazebosim/gazebo-classic #2866: After trying to save world from GUI, Gazebo freezes

gauravbhosale, 2020-10-27:
Basically after trying to save, Gazebo freezes, and this message appears in the terminal:
> GtkDialog mapped without a transient parent. This is discouraged.
[image]

This issue seen for Gazebo 9, 10 and 11 versions.
Is there any workaround for this?

chapulina, 2020-10-27:
I just tried this here, and I also see the message, but Gazebo is not frozen. How did you install Gazebo? Is there more information you can provide about your setup to help others reproduce it?

gauravbhosale, 2020-10-27:
1] I have used **Gazebo-Ubuntu installation** steps mentioned in the following page
http://gazebosim.org/tutorials?tut=install_ubuntu
The Gazebo installation is performed on **Ubuntu 18.04 VM**.

2] Further, I launched Gazebo using 'gazebo --verbose' command

3] After launching Gazebo, I tried to save world using Gazebo GUI options, **File->Save world as**

4] Once I clicked on 'Save world as', Gazebo freezes. In this state, the simulation time keep on increasing but I can not click on any object/button from Gazebo GUI. 
Also, 'Save world as' should provide option of saving world or selecting folder/path, but that didn't happen after click. 
So, basically, it look like freeze state or 'Save world as' is not working as expected.

SdKay, 2020-11-10:
I have same problem in a remote xrdp server and could be fix by running gazebo with sudo
But in my own computer it works well
But don't know how to fix this in other ways..becase not everyone have sudo permissions

Joggel1, 2021-01-05:
I have the same problem. Also as SdKay mentioned, it works when running gazebo with sudo. Interesting is, that the 'safe world as'-window appears, when I use ctrl+c (but I still can't save the world then):

https://user-images.githubusercontent.com/59475687/103637281-89532080-4f4b-11eb-88bd-798636ac2e63.mp4

BingyiSu, 2021-01-23:
I have the same problem and have you guys solved this problem?

Joggel1, 2021-01-30:
I accidently found out that if you minimze the gazebo window and maximize it again, the save-window appears, but after every action (e.g. click on a folder) I have to minimize and maximize again. So somehow this window just don't updates its view.

tim-fan, 2021-02-17:
I'm also seeing this, on Gazebo version 11.3.0 (on Ubuntu 20.04.2 LTS).

I've noted that the freeze only occurs when saving a world that contains one of my own custom models. In contrast I can save the world without issue if the world contains only models from models.gazebosim.org or fuel.ignitionrobotics.org.

This seems a little different from the issues described by @gauravbhosale and @Joggel1 , as they are not importing custom models. 

As per @Joggel1's suggestion, I can save the world successfully by repeatedly minimising then maximising before each click in the dialogue box.

liepgp, 2021-06-25:
I have the same problem
- Fresh installed Ubuntu 20.04.2 LTS in VirtualBox
- Gazebo installed from software center

Minimizing and and maximizing makes the save window to appear
Running with sudo, it works well

trejkev, 2021-09-01:
Just open it with sudo, e.g. 'sudo gazebo name_of_my_arena.world'. It worked pretty well for me using Ubuntu 18.

chongjeelee, 2021-11-12:
i also have the same problem,anyone solved this ?

jjbecomespheh, 2022-03-07:
> I accidently found out that if you minimze the gazebo window and maximize it again, the save-window appears, but after every action (e.g. click on a folder) I have to minimize and maximize again. So somehow this window just doesn't update its view.

This works for me, thanks so much @Joggel1 !

co4erol, 2022-06-11:
> Just open it with sudo, e.g. 'sudo gazebo name_of_my_arena.world'. It worked pretty well for me using Ubuntu 18.

you just save my life 👍

Aya-nuomi, 2022-06-21:
I've met another problem, when I saved the world use the method described above and try to load the world, gazebo can only show empty world without the model I saved in my_world.world.
Anyone met this?

yabdulra, 2022-06-30:
> I've met another problem, when I saved the world use the method described above and try to load the world, gazebo can only show empty world without the model I saved in my_world.world. Anyone met this?

What models are you importing to your world? I had similar problem and found out that it happened because I was using some models I found online. Worked well with models from gazebo online library and the ones I created myself.

ashwanthrkumar, 2023-11-05:
> I accidently found out that if you minimze the gazebo window and maximize it again, the save-window appears, but after every action (e.g. click on a folder) I have to minimize and maximize again. So somehow this window just doesn't update its view.

Thank you, issue cleared following this

ddowling, 2024-02-07:
I have found the same problem running Ubuntu 22 with a Gnome desktop. I seems to be related to the Gnome file manager. I found a workaround as follows:
```
XDG_CURRENT_DESKTOP=kde gazebo world.sdf
```
This presents a KDE style dialog for "Save As..." that seems to avoid the problem.

Kavinaya7, 2024-07-29:
Use Sudo gazebo, when starting gazebo