gazebosim/gazebo-classic #2867: TBB task is deprecated

traversaro, 2020-10-30:
While compiling gazebo with a recent version of tbb, I noticed this warning:
~~~
$PREFIX/include/tbb/task.h:21:139: note: #pragma message: TBB Warning: tbb/task.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
 #pragma message("TBB Warning: tbb/task.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
~~~

Indeed, apparently the tbb task has been deprecated, see https://github.com/oneapi-src/oneTBB/issues/243 for a related discussion. I did not find any info on whether or when the support will be actually removed.

huseyinafsin, 2021-02-18:
same problem

traversaro, 2021-03-03:
For make sure that this issue if found by people that may encounter this problem, this is the compilaton errors if Gazebo is compiled against tbb 2021.1 : 
~~~
2021-03-03T17:28:51.9367881Z In file included from ./gazebo/transport/transport.hh:3,
2021-03-03T17:28:51.9368421Z                  from ../gazebo/gazebo_client.cc:20:
2021-03-03T17:28:51.9369504Z ../gazebo/transport/Connection.hh:58:5: error: expected class-name before '{' token
2021-03-03T17:28:51.9370041Z    58 |     {
2021-03-03T17:28:51.9370295Z       |     ^
2021-03-03T17:28:51.9371095Z ../gazebo/transport/Connection.hh:73:20: error: 'task' in namespace 'tbb' does not name a type
2021-03-03T17:28:51.9371709Z    73 |       public: tbb::task *execute()
2021-03-03T17:28:51.9372071Z       |                    ^~~~
2021-03-03T17:28:51.9372757Z In file included from /usr/share/miniconda/envs/test/include/tbb/task.h:17,
2021-03-03T17:28:51.9373369Z                  from ../gazebo/transport/Connection.hh:20,
2021-03-03T17:28:51.9373926Z                  from ./gazebo/transport/transport.hh:3,
2021-03-03T17:28:51.9374416Z                  from ../gazebo/gazebo_client.cc:20:
2021-03-03T17:28:51.9375652Z /usr/share/miniconda/envs/test/include/oneapi/tbb/task.h:26:11: note: 'tbb::v1::task' declared here
2021-03-03T17:28:51.9376210Z    26 | namespace task {
2021-03-03T17:28:51.9376535Z       |           ^~~~
2021-03-03T17:28:51.9376972Z In file included from ./gazebo/transport/transport.hh:3,
2021-03-03T17:28:51.9377508Z                  from ../gazebo/gazebo_client.cc:20:
2021-03-03T17:28:51.9379076Z ../gazebo/transport/Connection.hh: In member function 'void gazebo::transport::Connection::OnReadData(const boost::system::error_code&, boost::tuples::tuple<Handler>)':
2021-03-03T17:28:51.9380432Z ../gazebo/transport/Connection.hh:313:61: error: 'allocate_root' is not a member of 'tbb::v1::task'
2021-03-03T17:28:51.9381153Z   313 |                   ConnectionReadTask *task = new(tbb::task::allocate_root())
2021-03-03T17:28:51.9381834Z       |                                                             ^~~~~~~~~~~~~
2021-03-03T17:28:51.9382671Z ../gazebo/transport/Connection.hh:315:30: error: 'enqueue' is not a member of 'tbb::v1::task'
2021-03-03T17:28:51.9383227Z   315 |                   tbb::task::enqueue(*task);
2021-03-03T17:28:51.9383581Z       |                              ^~~~~~~
2021-03-03T17:28:51.9384016Z In file included from ./gazebo/transport/transport.hh:6,
2021-03-03T17:28:51.9384538Z                  from ../gazebo/gazebo_client.cc:20:
2021-03-03T17:28:51.9384992Z ../gazebo/transport/Node.hh: At global scope:
2021-03-03T17:28:51.9385763Z ../gazebo/transport/Node.hh:40:5: error: expected class-name before '{' token
2021-03-03T17:28:51.9386221Z    40 |     {
2021-03-03T17:28:51.9386482Z       |     ^
2021-03-03T17:28:51.9387094Z ../gazebo/transport/Node.hh:54:20: error: 'task' in namespace 'tbb' does not name a type
2021-03-03T17:28:51.9387636Z    54 |       public: tbb::task *execute()
2021-03-03T17:28:51.9387986Z       |                    ^~~~
2021-03-03T17:28:51.9388440Z In file included from /usr/share/miniconda/envs/test/include/tbb/task.h:17,
2021-03-03T17:28:51.9389036Z                  from ../gazebo/transport/Connection.hh:20,
2021-03-03T17:28:51.9389539Z                  from ./gazebo/transport/transport.hh:3,
2021-03-03T17:28:51.9470385Z                  from ../gazebo/gazebo_client.cc:20:
2021-03-03T17:28:51.9471569Z /usr/share/miniconda/envs/test/include/oneapi/tbb/task.h:26:11: note: 'tbb::v1::task' declared here
2021-03-03T17:28:51.9472170Z    26 | namespace task {
2021-03-03T17:28:51.9472485Z       |           ^~~~
2021-03-03T17:28:51.9472931Z In file included from ./gazebo/transport/transport.hh:6,
2021-03-03T17:28:51.9473493Z                  from ../gazebo/gazebo_client.cc:20:
2021-03-03T17:28:51.9474996Z ../gazebo/transport/Node.hh: In member function 'void gazebo::transport::Node::Publish(const string&, const google::protobuf::Message&)':
2021-03-03T17:28:51.9476312Z ../gazebo/transport/Node.hh:162:52: error: 'allocate_root' is not a member of 'tbb::v1::task'
2021-03-03T17:28:51.9477006Z   162 |                 PublishTask *task = new(tbb::task::allocate_root())
2021-03-03T17:28:51.9477768Z       |                                                    ^~~~~~~~~~~~~
2021-03-03T17:28:51.9479221Z ../gazebo/transport/Node.hh:165:28: error: 'enqueue' is not a member of 'tbb::v1::task'
2021-03-03T17:28:51.9480034Z   165 |                 tbb::task::enqueue(*task);
2021-03-03T17:28:51.9480405Z       |                            ^~~~~~~
2021-03-03T17:28:53.7140198Z [343/1750] Building CXX object gazebo/CMakeFiles/libgazebo_client.dir/gazebo_shared.cc.o
2021-03-03T17:28:53.7142057Z FAILED: gazebo/CMakeFiles/libgazebo_client.dir/gazebo_shared.cc.o 
2021-03-03T17:28:53.7154404Z /usr/share/miniconda/envs/test/bin/x86_64-conda-linux-gnu-c++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBUILDING_DLL -DHAVE_OPENGL -DIGN_PROFILER_ENABLE=0 -DLIBBULLET_VERSION=0.0 -Dlibgazebo_client_EXPORTS -I../test/gtest/include -I../ -I. -isystem /usr/share/miniconda/envs/test/include/ignition/msgs5 -isystem /usr/share/miniconda/envs/test/include/ignition/math6 -isystem /usr/share/miniconda/envs/test/include/ignition/transport8 -isystem /usr/share/miniconda/envs/test/include/ignition/com [... cut ...]
2021-03-03T17:28:53.7204037Z       |                                                    ^~~~~~~~~~~~~
2021-03-03T17:28:53.7204800Z ../gazebo/transport/Node.hh:165:28: error: 'enqueue' is not a member of 'tbb::v1::task'
2021-03-03T17:28:53.7205299Z   165 |                 tbb::task::enqueue(*task);
2021-03-03T17:28:53.7205643Z       |                            ^~~~~~~
2021-03-03T17:28:54.8664323Z [344/1750] Building CXX object gazebo/CMakeFiles/gzserver.dir/server_main.cc.o
2021-03-03T17:28:54.8670881Z ninja: build stopped: subcommand failed.
 ~~~

traversaro, 2021-03-03:
Note that while in conda we can pin an old tbb version and I guess tbb will take a while to get propagated to stable Debian/Ubuntu release, this may be a more immediate problem if the tbb version in homebrew gets updated. fyi @scpeters

traversaro, 2021-03-03:
For an example of an "attempted" migration from `tbb::task`: https://github.com/borglab/gtsam/pull/336 .

scpeters, 2021-05-21:
as you predicted, this is now hitting us in homebrew: https://github.com/osrf/homebrew-simulation/issues/1477

given that `tbb::task` is part of our public API in `gazebo::transport::Connection`, we will have to use a pinned version of `tbb` unless we break API

https://github.com/osrf/gazebo/blob/gazebo11_11.5.1/gazebo/transport/Connection.hh#L57

scpeters, 2021-05-21:
> we will have to use a pinned version of `tbb` unless we break API

I think we could use `ifdef`s to allow a different API depending on the `tbb` version

scpeters, 2021-05-22:
I've managed to fix the gazebo11 homebrew bottle for now in https://github.com/osrf/homebrew-simulation/issues/1477 by using the `tbb@2020` formula, but that formula is marked as deprecated, so I expect it will be removed at some point:

* [image] https://build.osrfoundation.org/job/gazebo-install-one_liner-homebrew-amd64/2057/

~~~
+ brew doctor
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry or file an issue; just ignore this. Thanks!

Warning: Some installed formulae are deprecated or disabled.
You should find replacements for the following formulae:
  tbb@2020
~~~

scpeters, 2021-06-23:
there was an attempt to use the new API in #3027, but it doesn't compile

acxz, 2021-06-26:
Sorry about that failed PR, I decided to submit a PR to get people started on how to potentially fix the issue. If I get some time later, I'll investigate the compilation issues.

If someone else takes this up before me, here is a useful guide on migrating from the older deprecated low level `tbb::tasks` to the high level `tbb::task_groups` that use functors: https://docs.oneapi.com/versions/latest/onetbb/tbb_userguide/Migration_Guide/Task_API.html

scpeters, 2021-07-01:
> I've managed to fix the gazebo11 homebrew bottle for now in [osrf/homebrew-simulation#1477](https://github.com/osrf/homebrew-simulation/issues/1477) by using the `tbb@2020` formula, but that formula is marked as deprecated, so I expect it will be removed at some point:

since then, I have added a copy of `tbb@2020_u3` to `osrf/simulation` in https://github.com/osrf/homebrew-simulation/pull/1483. This required some extra logic to detect this version of `tbb` in https://github.com/ignition-tooling/release-tools/pull/466. Unfortunately that logic is now broken:

* [image] https://build.osrfoundation.org/job/gazebo-ci-gazebo11-homebrew-amd64/76/

I believe the issue is that a recent change to `tbb` in https://github.com/Homebrew/homebrew-core/pull/80329 has added a `.pc` file, so now the wrong version is being detected. I will see if we can add a version constraint to prevent finding versions that are too new.

FederAndInk, 2021-12-12:
after looking at #3027 and trying to compile it and pinpoint the issues:

First of all, easy to fix issue is an `operator()()` that shouldn't be const.

After removing the const, here is what I get:

```
[build] In file included from /usr/include/oneapi/tbb/detail/_small_object_pool.h:23,
[build]                  from /usr/include/oneapi/tbb/detail/_task.h:23,
[build]                  from /usr/include/oneapi/tbb/task_group.h:26,
[build]                  from /usr/include/tbb/task_group.h:17,
[build]                  from /home/nihil/Nextcloud/coding/cpp/gazebo/gazebo/transport/Connection.hh:20,
[build]                  from /home/nihil/Nextcloud/coding/cpp/gazebo/build/gazebo/transport/transport.hh:3,
[build]                  from /home/nihil/Nextcloud/coding/cpp/gazebo/gazebo/gazebo_core.hh:21,
[build]                  from /home/nihil/Nextcloud/coding/cpp/gazebo/gazebo/gazebo.hh:20,
[build]                  from /home/nihil/Nextcloud/coding/cpp/gazebo/build/plugins/rest_web/RestUiPlugin_autogen/EWIEGA46WW/../../../../../plugins/rest_web/RestUiWidget.hh:27,
[build]                  from /home/nihil/Nextcloud/coding/cpp/gazebo/build/plugins/rest_web/RestUiPlugin_autogen/EWIEGA46WW/moc_RestUiWidget.cpp:10,
[build]                  from /home/nihil/Nextcloud/coding/cpp/gazebo/build/plugins/rest_web/RestUiPlugin_autogen/mocs_compilation.cpp:3:
[build] /usr/include/oneapi/tbb/profiling.h:229:15: error: expected unqualified-id before ‘)’ token
[build]   229 |     void emit() { }
[build]       |               ^
[build] /usr/include/oneapi/tbb/profiling.h:231:22: error: expected unqualified-id before ‘const’
[build]   231 |     static void emit(const std::string &) { }
[build]       |                      ^~~~~
[build] /usr/include/oneapi/tbb/profiling.h:231:22: error: expected ‘)’ before ‘const’
[build]   231 |     static void emit(const std::string &) { }
[build]       |                     ~^~~~~
[build]       |                      )
```

the problem here is that `emit` is a macro defined by qt and here defined by tbb :/

I'll be doing further investigations, soon.

FederAndInk, 2021-12-12:
So, to make it compile, we can encapsulate `<tbb/task_group>` in a header and include that instead like that:

for instance in file gazebo/util/tbb_task_group.hh:

```cpp
#ifndef _GAZEBO_UTIL_TBB_TASK_GROUP_HH_
#define _GAZEBO_UTIL_TBB_TASK_GROUP_HH_
#ifdef emit
#undef emit
#include <tbb/task_group.h>
#define emit
#else
#include <tbb/task_group.h>
#endif
#endif
```

FederAndInk, 2021-12-12:
After looking a bit more at the code, here are 2 things that I think we need to consider:

first, the 3 tasks created seem to be used on code that is waiting (for data, ...), here it is advised against using tasks for these use cases: https://www.intel.com/content/dam/develop/external/us/en/documents/1-6-appthr-using-tasks-instead-of-threads-165657.pdf
-> maybe we could use threads instead

then, the `task_group` class is managing it's tasks and we have to call `wait` on it, we could add a static member `task_group` to the concerned classes (or create a singleton?)

I don't know the code of gazebo in details, so comments are welcomed :)

commit ea956014 (2022-03-10): Replace deprecated tbb task for tbb >= 2021 (#3174)

As tbb::task has been removed in oneTBB 2021.01,
we need to replace its use with oneapi::tbb::task_group.
Define a wrapper so that tbb::task_group is used for
newer versions of oneTBB.

Fixes #2867.

* Fix typo
* Remove TopicManagerProcessTask as it is unused
* Use oneapi::tbb::task_group instead of tbb::task if available
* Assign my copyright to OSRF
* Add myself to AUTHORS
* Default to using tbb::task for older versions of TBB, to maintain ABI
* Replace use of tbb/version.h with tbb/tbb.h

As tbb/version.h is seemingly not present before v [... cut]
closed 2022-03-10 by scpeters