gazebosim/gazebo-classic #2869: Cannot spawn robot from urdf FrameAttachedToGraph Error

ddonatien, 2020-11-02:
Hello,

I get the following error while spawning a robot in gazebo using gazebo_ros spawn_model :

```
[Err] [Model.cc:99] Error Code 23 Msg: FrameAttachedToGraph error, Non-LINK vertex with name [__model__] is disconnected; it should have 1 outgoing edge in MODEL attached_to graph.
[Err] [Model.cc:99] Error Code 23 Msg: Graph has __model__ scope but sink vertex named [__model__] does not have FrameType LINK when starting from vertex with name [__model__].
```

The urdf is from franka_ros and check_urdf does not return any error.

I am using Gazebo 11.1.0 with Ros Noetic on Ubuntu 20.04.1 and the following launch file :
```
<?xml version="1.0" encoding="utf-8"?>
<launch>
  <!-- Load rosparams -->
  <param name="robot_description" command="$(find xacro)/xacro $(find franka_description)/robots/panda_arm.urdf.xacro"/>

  <!-- Launch Gazebo -->
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="verbose" value="true"/>
  </include>

  <!-- Spawn urdf of the robot -->
  <node name="robot_description" pkg="gazebo_ros" type="spawn_model" args="-param robot_description -urdf -model panda"/>
</launch>
```
I don't have any issue spawning the tricycle from gazebo_plugin, I don't understand one works and the other doesn't.

chapulina, 2020-11-05:
It would be helpful to have the URDF file you're using. Can you run `gz sdf --check` on the file to see if there are any errors?

ddonatien, 2020-11-05:
The urdf is from https://github.com/frankaemika/franka_ros/blob/noetic-devel/franka_description/robots/panda_arm.xacro
The urdf from xacro is the following : https://gist.github.com/ddonatien/9b2514ba6d5c9d83f511713aedbed5b0
gz sdf --check returns :
`Check complete`

scpeters, 2020-11-06:
running `ign sdf -p panda_arm.urdf` gives the following:

~~~
<sdf version='1.7'>
  <model name='panda'/>
</sdf>
~~~

I looked in the debug log at `~/.sdformat/sdformat.log`:

~~~
Dbg [parser_urdf.cc:422] Fixed Joint Reduction: extension lumping from [panda_link8] to [panda_link7]
Dbg [parser_urdf.cc:1007] lumping collision [] for link [panda_link8] to parent [panda_link7] with name [panda_link8]
Dbg [parser_urdf.cc:1007] lumping collision [] for link [panda_link8] to parent [panda_link7] with name [panda_link8]
Dbg [parser_urdf.cc:1007] lumping collision [] for link [panda_link8] to parent [panda_link7] with name [panda_link8]
Dbg [parser_urdf.cc:2662] urdf2sdf: link[panda_link0] has no inertia, [1] children links ignored.
Dbg [parser_urdf.cc:2670] urdf2sdf: link[panda_link0] has no inertia, [1] children joints ignored.
Dbg [parser_urdf.cc:2684] urdf2sdf: link[panda_link0] has no inertia, not modeled in sdf
Dbg [parser.cc:420] parse from urdf file [/tmp/panda_arm.urdf].
~~~

so it looks like it wants `panda_link0` to have inertia parameters specified. we should probably improve our console messages

scpeters, 2020-11-06:
https://github.com/osrf/sdformat/issues/199

ddonatien, 2020-11-06:
I checked, adding inertia to the links solves the issue. Thank you for your help, I think I can close this issue now !

closed 2020-11-06 by ddonatien