gazebosim/gazebo-classic #2872: ubuntu 20.04 gazebo 11.0.0 Unregistered physics engine [dart]

ycpss91255, 2020-11-06:
surroundings:
  * osrf/ros:noetic-desktop-full
  * gazebo 11.0.0

error code:
  [Err] [Server.cc:442] Unregistered physics engine [dart], the default will be used instead.

I use osrf/ros:noetic-desktop-full without dart physics engine when executing gazebo
[image]

chapulina, 2020-11-07:
Could you try updating your installation? I believe DART support is shipped with Gazebo since version 11.1.0, see https://github.com/osrf/gazebo/issues/2752.

ycpss91255, 2020-11-07:
Ok, Let me try.

ycpss91255, 2020-11-09:
Hi! I succeeded. I use orsf/ros-noetic-desktop-focal and sudo apt-get install gazebo 11, but I want to ask if this bug will be fixed?

chapulina, 2020-11-09:
I'm glad to hear it worked. It looks like there isn't a bug, the issue has been addressed in a newer version, so I'm closing this issue.

closed 2020-11-09 by chapulina

ycpss91255, 2020-11-10:
I found that simply sudo apt-get install gazebo11 can also solve the problem