gazebosim/gazebo-classic #2876: Ubuntu .deb package is not found

elpiel, 2020-11-07:
Hello,
I tried downloading the `.deb` package however there is no such file:

1. Open http://gazebosim.org/download
2. Debian `.deb`
3. Download for Ubuntu
4. You get redirected to http://osrf-distributions.s3.amazonaws.com/gazebo/gazebo.sh which shows `NoSuchKey` error

The link (http://get.gazebosim.org/) in the text `Download gazebo.sh script, and run`  works

I couldn't find a repository for the website, so if I've posted it in the wrong repository - Sorry :sweat_smile:

scpeters, 2020-11-18:
it looks like the blue `Download` button has an invalid link. thanks I'll figure out where the code is to fix this

whiskygrandee, 2021-02-17:
I checked and the issue hasn't been fixed yet. Please do it, I have an assignment on this.

closed 2021-05-14 by chapulina

chapulina, 2021-05-14:
The button is fixed now.