gazebosim/gazebo-classic #2883: PluginTest failing since plugins became static libraries

chapulina, 2020-11-19:
These tests started failing recently:

* `PluginTest.ModelExceptionConstructor`
* `PluginTest.ModelExceptionInit`
* `PluginTest.ModelExceptionLoad`

I can reproduce the failures locally. The problem is that the plugins can't be loaded anymore. Here's the test log:

```
[Err] [Plugin.hh:178] Failed to load plugin libExceptionModelPluginConstructor.so: libExceptionModelPluginConstructor.so: cannot open shared
 object file: No such file or directory                                                                                                     
...
/home/chapulina/gazebo11_ws/src/gazebo/test/integration/plugin.cc:58: Failure                                  
Value of: loggedString.find( "Exception occured in the constructor of plugin with name") != std::string::npos
  Actual: false                                                                         
Expected: true
```

Changing the libraries to `SHARED` here fixes the tests for me:

https://github.com/osrf/gazebo/blob/5ecd18aead9cf70574486979b161ad5c12a5faa2/test/plugins/CMakeLists.txt#L34

This was changed to `STATIC` on #2753 and #2754.

@j-rivero , do you think we can revert that one change? Or maybe fix the test some other way? Right now the plugins have `.a` extensions and the test is expecting `.so`.

scpeters, 2020-12-03:
yeah, building a plugin as `STATIC` doesn't make any sense to me

j-rivero, 2020-12-04:
god, that makes no sense indeed.  I'll send the PR to fix it.

PR #2897 (2021-05-08): Compile test plugins as MODULE instead of STATIC
Revert part of #2753 which change the compilation of plugins to STATIC.

Should fix #2883 in the gazebo11 branch. Needs to be tested on ARM.
closed 2021-05-08 by scpeters