gazebosim/gazebo-classic #2884: DEM origin overrides spherical coordinate flags defined in .world

Jaeyoung-Lim, 2020-11-20:
**Problem Description**
When using DEM models that have the global origin in the meta data, it is not possible to override the world location using the `spherical_coordinates` tag in the sdf file(world).

Even if the `spherical_coordinates` are modified, the `world->SphericalCoords()` returns the origin of the DEM file, not the origin defined with the `spherical_coordinates` flag.

It seems like gazebo overrides the spherical coordinate defined in the world file and uses the origin of the DEM if I understand correctly the following.

https://github.com/osrf/gazebo/blob/063d7386ed5499c2883de48e19ff5fb9dec01dff/gazebo/physics/HeightmapShape.cc#L109-L131
```
    // Modify the reference geotedic latitude/longitude.
    // A GPS sensor will use the real georeferenced coordinates of the terrain.
    common::SphericalCoordinatesPtr sphericalCoordinates;
    sphericalCoordinates = this->world->SphericalCoords();

    if (sphericalCoordinates)
    {
      ignition::math::Angle latitude, longitude;
      double elevation;

      this->dem.GetGeoReferenceOrigin(latitude, longitude);
      elevation = this->dem.GetElevation(0.0, 0.0);

      sphericalCoordinates->SetLatitudeReference(latitude);
      sphericalCoordinates->SetLongitudeReference(longitude);
      sphericalCoordinates->SetElevationReference(elevation);
      sphericalCoordinates.reset();
    }
    else
      gzerr << "Unable to get a valid SphericalCoordinates pointer\n";

    return 0;
  }
```

**Expected behavior**
- `world->SphericalCoords()` returns the origin of the DEM file if there are no `spherical_coordinates` defined in the world file
- `world->SphericalCoords()` returns the location defined in `spherical_coordinates` if defined in world file

@ahcorde FYI

**Additional Context**
- This issue was also posted in https://github.com/osrf/gazebo_models/issues/43, but cross posting since there was no response

Jaeyoung-Lim, 2021-01-16:
@chapulina @ahcorde @tfoote Would any one know how to move forward regarding this issue?

Vicidel, 2023-07-18:
Hello all. I know this issue is quite old, but the bug seems to be still happening. See for example [this issue in PX4 forum](https://discuss.px4.io/t/px4-home-lat-and-px4-home-lon-not-correct-when-using-a-heightmap-in-gazebo/32927).
Pinging recent contributors @chapulina @scpeters @iche033, anything you can do would help me.

tfoote, 2023-08-09:
The best way to push this forward would be to open a PR with the expected behavior.