gazebosim/gazebo-classic #2893: Gazebo & SpaceNav Pro crash

ViaudJV, 2020-11-27:
Hi,
I work with gazebo, i see a crash of gazebo-9 with spaceNav Pro.
This spacemouse have more than two buttons (14 bouttons).
The spacemouse event return a sev.button.bnum with a value up to 26.
I think to resolve the crash i suggest to increase number of button definition on SpaceNavPrivate.hh and modify SpaceNav.cc
or check sev.button.bnum value before affectation.

chapulina, 2020-11-30:
Thank you for the bug report. You're welcome to open a pull request with the suggested fix. We won't have the hardware to test with, but if the code looks reasonable and maintains backwards compatibility I think we can get it in.

chapulina, 2020-11-30:
And I just saw your pull request 20 seconds after writing that reply :smile: 

https://github.com/osrf/gazebo/pull/2895/