gazebosim/gazebo-classic #2899: Gazebo on Windows: conda-forge installation

j-rivero, 2020-12-09:
The [Conda](https://docs.conda.io/en/latest/) project has available a package for [Gazebo11 inside the conda-forge](https://anaconda.org/conda-forge/gazebo) organization. The Goal of the ticket is to evaluate if this installation is stable enough to go into our documentation.

**Help wanted:**
We are interested in knowing if our users on Windows can install and use it without problems. If you want to help, please follow Conda instructions to install anaconda, miniconda or other variant of it, try to install Gazebo from conda-forge and try to run a basic simulation on it. Please comment in the issue with details about Windows version, conda installer, problems/not-problems found, graphic card drivers installed and other information you find useful. You are also welcome to comment and help with other people's problem.

**Windows specific:**
 * Be careful with the firewall rules
 * Be careful with Antivirus actions interfering Gazebo run

j-rivero, 2020-12-09:
My testing of the superbuild:
 * **System**: Win10 Home 64bits
 * **Graphics**: Geforce MX150 + Intel (configured in nvidia control panel to run gzserver/gzclient on nvidia), driver version 398.36, DirectX runtime 12.
 * **Conda version**: Miniconda installation, gazebo 11.3.0 package used from conda-forge 
 * **Other**: Disabled antivirus.

**Runtime testing**:
```batch
conda install gazebo
...
C:\ProgramData\Miniconda3\Library\share\gazebo\setup.bat
gazebo --verbose
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
```
seems like something is blocking the network connection or the network connection is miss-configured somehow. The GUI did not appear. Did not have time to debug this.

traversaro, 2020-12-09:
Which shell are you using? Powershell and Git Bash are currently know to not be working (see https://github.com/conda-forge/gazebo-feedstock/issues/42).  Note that it should not be necessary to call `C:\ProgramData\Miniconda3\Library\share\gazebo\setup.bat` manually, but it should be sufficient just to re-open a new command prompt and just activate conda again there. 

Note that it seems that you installed Gazebo in the base environment, it is generally recommended to avoid to install heavy packages in the base environment, but rather to create a new environment and install the dependencies that you want to use there. 

cc @wolfv @Tobias-Fischer

j-rivero, 2020-12-11:
> Which shell are you using?

plain cmd.exe. 

> Note that it seems that you installed Gazebo in the base environment, it is generally recommended to avoid to install heavy packages in the base environment, but rather to create a new environment and install the dependencies that you want to use there.

Thanks for the tip. Seems to me like it should not change the behavior of the problem, is this correct?

wolfv, 2020-12-11:
@j-rivero it would be interesting to know which version of gazebo got installed, too.

wolfv, 2020-12-11:
PS: this is exciting! I am happy to help in any way I can!

j-rivero, 2020-12-14:
Thanks @wolfv ! From my previous comment:

>     * **Conda version**: Miniconda installation, gazebo 11.3.0 package used from conda-forge

danzimmerman, 2020-12-29:
```
Windows Insider Build 20279.fe_release.201209-1332
miniconda3 
conda 4.8.4
NVIDIA GTX 1050 Ti / Driver version 460.20
```

Installed with 

```
conda install gazebo -c conda-forge
```
into a virtual environment (Python 3.8 if that matters)

Set `gzserver.exe` and `gzclient.exe` to use the NVIDIA card in NVIDIA control panel, and to allow comms through the firewall:

[image]

Ran 

```
miniconda3\envs\<envname>\Library\share\gazebo\setup.bat
```

## Expected Behavior

The command
```
gazebo --verbose
```
should show some messages and then pop up some GUI windows (I guess? I am new to Gazebo.)

## Actual Behavior
Running from an Anaconda prompt for the virtual environment, after running `setup.bat`:

```
(<envname>) C:\Code>gazebo --verbose
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.140
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.140

<about three seconds delay>

Silently dies, no messages, no windows...
```

Same behavior with Antivirus disabled. Anything else I should try?

traversaro, 2020-12-29:
> Please let me know if this is useful or too n00b-y to be helpful. I don't have a clear entry point into this, just found it trying to find Gazebo binary builds for Windows and thought I'd give it a try.

Thanks @danzimmerman for the useful feedback! I personally think that any feedback is useful, until gazebo runs easily on Windows. Two additional questions: 
* Can you copy&paste the output of the `conda list` and `conda info`  commands? Furthermore, to understand if there is anything that is interfering with your environment, it would be useful to know the output of `set`  command in the command prompt terminal with the gazebo environment enabled. 
* I read that both you and @j-rivero manually ran the `setup.bat`  script, however this should not be necessary as the `setup.bat`  script is already run by the activation process of the environment. 
* What happens if you try to run `gzserver.exe --verbose`  in one terminal and `gzclient --verbose`  in another? 
* What happens if you don't specify configure `gzserver.exe`  and `gzclient.exe`

traversaro, 2020-12-29:
> Thanks for the tip. Seems to me like it should not change the behavior of the problem, is this correct?

Sorry, I did not notice this message. In theory not, but if there is some bug in environment variables handling somewhere that affects the behavior and cause the crash.

danzimmerman, 2020-12-29:
Output of `conda info`:

```
     active environment : npe
    active env location : C:\Users\<username>\miniconda3\envs\npe
            shell level : 1
       user config file : C:\Users\<username>\.condarc
 populated config files : 
          conda version : 4.8.4
    conda-build version : not installed
         python version : 3.8.3.final.0
       virtual packages : __cuda=11.2
       base environment : C:\Users\<username>\miniconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\<username>\miniconda3\pkgs
                          C:\Users\<username>\.conda\pkgs
                          C:\Users\<username>\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\<username>\miniconda3\envs
                          C:\Users\<username>\.conda\envs
                          C:\Users\<username>\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.8.4 requests/2.24.0 CPython/3.8.3 Windows/10 Windows/10.0.20279
          administrator : False
             netrc file : None
           offline mode : False

```

Output of `conda list`:

```
# packages in environment at C:\Users\dan\miniconda3\envs\npe:
#
# Name                    Version                   Build  Channel
argon2-cffi               20.1.0           py38he774522_1  
astroid                   2.4.2                    py38_0  
attrs                     20.1.0                     py_0  
backcall                  0.2.0                      py_0  
blas                      1.0                         mkl  
bleach                    3.1.5                      py_0  
boost-cpp                 1.74.0               h54f0996_1    conda-forge
bqplot                    0.12.15            pyh9f0ad1d_0    conda-forge
brotlipy                  0.7.0           py38he774522_1000  
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2020.12.5            h5b45459_0    conda-forge
casadi                    3.5.3                    pypi_0    pypi
certifi                   2020.12.5        py38haa244fe_0    conda-forge
cffi                      1.14.2           py38h7a1dbc1_0  
chardet                   3.0.4                 py38_1003  
click                     7.1.2                      py_0  
cloudpickle               1.6.0                      py_0  
colorama                  0.4.3                      py_0  
console_bridge            1.0.1                h7ef1ec2_0    conda-forge
cppzmq                    4.7.1                h276ee43_0    conda-forge
cryptography              3.1              py38h7a1dbc1_0  
curl                      7.71.1               h4b64cdc_8    conda-forge
cycler                    0.10.0                   py38_0  
cytoolz                   0.10.1           py38he774522_0  
dash                      1.4.1                      py_0  
dash-core-components      1.3.1                      py_0  
dash-html-components      1.0.1                      py_0  
dash-renderer             1.1.2                      py_0  
dash-table                4.4.1                      py_0  
dask-core                 2.25.0                     py_0  
decorator                 4.4.2                      py_0  
[... 228 lines cut ...]
<re-starts gzserver in other window>

[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.140

<dies when gzserver dies>
```

> What happens if you don't specify configure gzserver.exe and gzclient.exe

You mean in the NVIDIA control panel? Same. Firewall rules were actually added when I first ran gazebo, okayed them in GUI window. 

Are there dependencies to gazebo conda-forge installation that conda wouldn't handle? I might be missing something? 

Also happy to create a cleaner environment to try this. I also don't know exactly how I'm going to interact with Gazebo. I've set up ROS melodic on Windows according to http://wiki.ros.org/Installation/Windows, using customized repos from https://ms-iot.github.io/ROSOnWindows/ where necessary.

**Aaaand... apparently, those Windows instructions mean I _already_ have a working installation of Gazebo 9.13.1 on my Windows machine.**

That gazebo installation seems to be built as a DLL and only can be accessed using `roslaunch`, so I can't test any executables, but that does work:

[image]

This was not really clear from my reading of the `ms-iot` stuff.

I expect that interacting with Gazebo via ROS is probably the way to go on Windows, so maybe I personally don't actually need to get this conda installation working. However, I'm happy to answer more questions.

danzimmerman, 2020-12-30:
Sorry forgot the `set` output. Slightly redacted:

```
set
ACCEPT_EULA=Y
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\<username>\AppData\Roaming
ChocolateyInstall=C:\ProgramData\chocolatey
ChocolateyLastPathUpdate=132536388273435521
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=<THE-COMPUTER-NAME>
ComSpec=C:\WINDOWS\system32\cmd.exe
CONDA_DEFAULT_ENV=npe
CONDA_EXE=C:\Users\<username>\miniconda3\Scripts\conda.exe
CONDA_PREFIX=C:\Users\<username>\miniconda3\envs\npe
CONDA_PROMPT_MODIFIER=(npe)
CONDA_PYTHON_EXE=C:\Users\<username>\miniconda3\python.exe
CONDA_SHLVL=1
configsetroot=C:\WINDOWS\ConfigSetRoot
DriverData=C:\Windows\System32\Drivers\DriverData
DXROOT=C:\Program Files (x86)\Sandcastle\
GAZEBO_MASTER_URI=http://localhost:11345
GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
GAZEBO_MODEL_PATH=C:/Users/<username>/miniconda3/envs/npe\Library\share\gazebo-11\models;
GAZEBO_PLUGIN_PATH=C:/Users/<username>/miniconda3/envs/npe\Library\bin\gazebo-11\plugins;
GAZEBO_RESOURCE_PATH=C:/Users/<username>/miniconda3/envs/npe\Library\share\gazebo-11;
GEM_HOME=C:\Users\<username>\miniconda3\envs\npe\Library\share\rubygems
GIT_LFS_PATH=C:\Program Files\Git LFS
HOME=C:\Users\<username>
HOMEDRIVE=C:
HOMEPATH=\Users\<username>
LD_LIBRARY_PATH=;C:/Users/<username>/miniconda3/envs/npe\Library\bin\gazebo-11\plugins
LOCALAPPDATA=C:\Users\<username>\AppData\Local
LOGONSERVER=\\<THE-COMPUTER-NAME>
MSMPI_BIN=C:\Program Files\Microsoft MPI\Bin\
NUMBER_OF_PROCESSORS=8
OGRE_RESOURCE_PATH=C:/Users/<username>/miniconda3/envs/npe\Library\bin
OneDrive=C:\Users\<username>\OneDrive
OneDriveConsumer=C:\Users\<username>\OneDrive
OS=Windows_NT
Path=C:\Users\<username>\miniconda3\envs\npe\Library\share\rubygems\bin;;C:/Users/<username>/miniconda3/envs/npe\Library\bin\gazebo-11\plugins;C:\Users\<username>\miniconda3\envs\npe;C:\Users\<username>\miniconda3\envs\npe\Library\mingw-w64\bin;C:\Users\<username>\miniconda3\envs\npe\Library\usr\bin;C:\Users\<username>\miniconda3\envs\npe\Library\bin;C:\Users\<username>\miniconda3\envs\npe\Scripts;C:\Users\<username>\miniconda3\envs\npe\bin;C:\Users\<username>\miniconda3\condabin;C:\Program Files\Microsoft MPI\Bin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\iCLS;C:\Program Files\Intel\Intel(R) Management Engine Components\iCLS;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\Windows\System32\OpenSSH;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Git LFS;C:\Users\les\AppData\Local\Programs\Python\Python37;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\WINDOWS\System32\OpenSSH;C;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Git\cmd;C:\ProgramData\chocolatey\bin;C:\Users\<username>\AppData\Local\Microsoft\WindowsApps;C:\Users\<username>\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\<username>\bin\ffmpeg-20200429-280383a-win64-static\bin;C:\Users\<username>\bin\Graphviz\bin;C:\Users\<username>\bin\swigwin-4.0.2;C:\Users\<username>\AppData\Local\GitHubDesktop\bin;C:\Users\<username>\bin\iperf;.
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 158 Stepping 10, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=9e0a
ProgramData=C:\ProgramData
ProgramFiles=C:\Program Files
ProgramFiles(x86)=C:\Program Files (x86)
ProgramW6432=C:\Program Files
PROMPT=(npe) $P$G
PSModulePath=C:\Program Files\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules
PUBLIC=C:\Users\Public
SDF_PATH=C:\Users\<username>\miniconda3\envs\npe\Library\share\sdformat9\1.7
SESSIONNAME=Console
SystemDrive=C:
SystemRoot=C:\WINDOWS
TEMP=C:\Users\<username>\AppData\Local\Temp
TMP=C:\Users\<username>\AppData\Local\Temp
USERDOMAIN=<THE-COMPUTER-NAME>
USERDOMAIN_ROAMINGPROFILE=<THE-COMPUTER-NAME>
USERNAME=<username>
USERPROFILE=C:\Users\<username>
VBOX_MSI_INSTALL_PATH=C:\Program Files\Oracle\VirtualBox\
VCPKG_DEFAULT_TRIPLET=x64-windows
VS140COMNTOOLS=C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\Tools\
windir=C:\WINDOWS
```
[... 86 lines cut ...]
`gzclient` pops up a splash so it runs but hangs at "Preparing your world..."

--- 

From a raw cmd, after activating the virtual env:

```
(npe) C:\Users\<username>\miniconda3\condabin>gazebo --verbose
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.

<hangs indefinitely here>
```

I'll try this on my personal laptop which I haven't tinkered with nearly as much with system-level tools/env variables. This feels like a mess.

danzimmerman, 2020-12-30:
Tried this on my personal laptop (`Windows 10 Home Ver 2004 Build 19041.685`, `Dell XPS13 9300`) with two different environments:
 * the `npe` environment, also Python 3.8.3. I've manually installed a few packages on the work laptop, so this environment is no longer identical to the env on my work laptop, but it should be very similar.
 * a fresh `gaz383` environment initialized minimally with Python 3.8.3 to match the `npe` environment.
 * a clone of `npe`, `npe-gaz`, into which I installed some packages that weren't in `npe` after Gazebo install but that were in `gaz383`.

I've attached the output of `set` and `conda list` for both environments.

## Fresh Environment (Works) vs. Existing Environment (Fails) - Anaconda Prompt

`gazebo --verbose` works perfectly fine from **Anaconda Prompt** in the fresh `gaz383` environment. The failures in the `npe` environment are the same as described above on my work laptop. Compared with the working `gaz383` environment, it looks like my `npe` environment is missing:

```
ilmbase
openexr
pugixml
zziplib
```

Attempting manual install of those four dependencies into a clone of the `npe` environment (`npe-gaz`) does not fix the issue.

I've distilled the differences between `gaz383` and my `npe-gaz` clone with the four added dependencies here:

[gaz383-npe-gaz_package_diffs.txt](https://github.com/osrf/gazebo/files/5752974/gaz383-npe-gaz_package_diffs.txt)

and for the original difference between `gaz383` and `npe` here:

[npe-gaz383-package-differences.txt](https://github.com/osrf/gazebo/files/5752955/npe-gaz383-package-differences.txt)

## Files

### Environment vars

I'm not really seeing differences in the environment vars:

[gaz383-envs.txt](https://github.com/osrf/gazebo/files/5752925/gaz383-envs.txt) (WORKS)
[npe-envs.txt](https://github.com/osrf/gazebo/files/5752929/npe-envs.txt) (FAILS, MISSING PKGS)
[npe-gaz-envs.txt](https://github.com/osrf/gazebo/files/5752985/npe-gaz-envs.txt) (ADDED PKGS, FAILS)

### `conda list` outputs
[gaz383-condalist.txt](https://github.com/osrf/gazebo/files/5752926/gaz383-condalist.txt) (WORKS)
[npe-condalist.txt](https://github.com/osrf/gazebo/files/5752930/npe-condalist.txt) (FAILS, MISSING PKGS)
[npe-gaz-condalist.txt](https://github.com/osrf/gazebo/files/5752978/npe-gaz-condalist.txt) (ADDED PKGS, FAILS)

---
## Summary

 * Installation with `conda` and use from Anaconda Prompt in a fresh, empty environment works fine.
 * Installation into an existing environment does not run, some packages are missing.
 * Adding those missing packages manually does not fix the broken environment. Presumably some dependencies are still wrong versions, etc.

traversaro, 2020-12-30:
> Adding those missing packages manually does not fix the broken environment. Presumably some dependencies are still wrong versions, etc.

By inspecting manually the `npe-gaz-condalist.txt` and `gaz383-condalist.txt`, the most striking difference is that the Ogre version is `1.10.12              hbcc8020_2` for `npe-gaz` (not working), while for `gaz383` is  it is `1.10.12              h71cedee_6`. The version is  the same, but what is relevant here is the last part, that is the build number (the one that in the recipe is set in https://github.com/conda-forge/ogre-feedstock/blob/1.10/recipe/meta.yaml#L16). The build 2 of the `1.10.2` version of ogre is quite old (1 year and 1 month ago, according to https://anaconda.org/conda-forge/ogre/files) and does not contain important bug fixes such as https://github.com/conda-forge/ogre-feedstock/pull/23 .  

I am not sure if we could debug in some way why  `1.10.12              hbcc8020_2`  was installed (@wolfv may know some tricks on how to do that why some conda/mamba command), but perhaps one strategy is try to run `conda update ogre` in `npe-gaz` and check if ogre gets updated to the build 6, and if not probably some message should be printed. 

Something that I did not think about is that probably instead of saving the `conda list` output, it could make sense to save an environment.yaml file (with `conda env export > environment.yml`, see https://docs.conda.io/projects/conda/en/latest/user-guide/tasks/manage-environments.html#sharing-an-environment) to easily reproduce the issues. 

> I can't add models with the Insert tab, but maybe this is user error?

Is this happening also with the Gazebo installed by ROSOnWindows? In that case, I suggest to open a new issue in this repo as it is probably a Gazebo upstream problem. If instead if only happens with the conda binaries, please open an issue in https://github.com/conda-forge/gazebo-feedstock, thanks!

danzimmerman, 2020-12-30:
> Something that I did not think about is that probably instead of saving the conda list output, it could make sense to save an environment.yaml file

Yep, done below. I can't attach YAML directly, so I zipped all the referenced files:
[gzdbg-danzimmerman-2020-12-30.zip](https://github.com/osrf/gazebo/files/5754909/gzdbg-danzimmerman-2020-12-30.zip)

I'm back on the work laptop starting fresh.

I can reproduce this issue with a simple starting `.yml` file. Creating this environment fails to solve for a working Gazebo binary:

```
name: gz-simple
channels:
  - defaults
  - conda-forge
dependencies:
  - python==3.8.3
  - gazebo
```
I can then verify that the installed OGRE build is at fault. Updating it explicitly to build `h71cedee_6` allows Gazebo to run. This fix also works in my complex `npe` environment.

# Minimal Reproduction

## `gz-postinstall`
 * Created, empty, from CLI with `conda create --name gz-postinstall python==3.8.3 `
 * Activated `gz-postinstall` and installed Gazebo with `conda install -c conda-forge gazebo`
 * WORKS: This results in a running Gazebo installation, model-add issues aside. `OGRE` is Build 6.
 * Exported to `gz-postinstall-env-solved.yml`

## `gz-simple`
 * Created from `gz-simple.yml`.
 * Activated and ran `gazebo --verbose`
 * FAILS: Gazebo dies silently, as expected, same as `npe` environment, etc. `OGRE` is Build 2.
 * Exported to `gz-simple-env-solved.yml`

## `gz-simple-clone`
 * Cloned `gz-simple` with `conda create --clone gz-simple --name gz-simple-clone`
 * Activated and ran `gazebo --verbose`
 * FAILS: Gazebo dies silently, as expected.
 * Exported to `gz-simple-clone-preogre-env-solved.yml`
 * Ran `conda update ogre` -> Says `# All requested packages already installed`
 * Ran `conda install ogre=1.10.12=h71cedee_6 -c conda-forge`
 * WORKS: This results in a running Gazebo installation, same model-add issues as `gz-postinstall`.
 * Exported to `gz-simple-clone-postogre-env-solved.yml`

danzimmerman, 2020-12-30:
> Is this happening also with the Gazebo installed by ROSOnWindows? In that case, I suggest to open a new issue in this repo as it is probably a Gazebo upstream problem. If instead if only happens [...]

They both fail, but differently, and ROS on Windows Gazebo 9.13.1 has a simple fix. It was failing with `tar: could not chdir to 'C:\Users\<username>/.gazebo/models'` and adding an empty `models` subdir to `'C:\Users\<username>/.gazebo` fixed it.

Perhaps related to 
 * https://github.com/ms-iot/ROSOnWindows/issues/90 
 * https://github.com/osrf/gazebo/issues/2611
 * https://github.com/osrf/gazebo/pull/2789
 
--- 

The Conda binary Gazebo 11 still fails on adding models from [models.gazebosim.org](https://models.gazebosim.org) with

```
[Err] [..\gazebo\common\ModelDatabase.cc:607] Missing model.config for model ""
[Err] [..\gazebo\common\ModelDatabase.cc:671] Invalid model manifest file[""]
```

And with [fuel.ignitionrobotics.org](https://fuel.ignitionrobotics.org) models with:
```
[Err] [..\gazebo\common\FuelModelDatabase.cc:139] Missing model.config for model
```
I'll open a `gazebo-feedstock` issue for that.

traversaro, 2020-12-30:
Thanks a lot @danzimmerman ! I think we are indeed reaching the core of the issue. Given that the issue is related to how the environment is created from a yaml file, could you also report the version of conda you have in the base environment?

danzimmerman, 2020-12-30:
> Thanks a lot @danzimmerman ! I think we are indeed reaching the core of the issue. Given that the issue is related to how the environment is created from a yaml file, could you also report the version of conda you have in the base environment?

The tests above were run with `conda 4.8.4`.

I updated the base environment `conda 4.9.2` and it solves for the exact same set of packages. Installed with `4.9.2`:
[gz-simple-condaupdate.yml-env-solved.txt](https://github.com/osrf/gazebo/files/5755175/gz-simple-condaupdate.yml-env-solved.txt)

`mamba 0.7.6` gets it right, however. Deleted and re-created `gz-simple` with `mamba env create --file gz-simple.yml`:
[gz-simple-env-mambasolved.yml.txt](https://github.com/osrf/gazebo/files/5755200/gz-simple-env-mambasolved.yml.txt)

wolfv, 2020-12-30:
thanks a million for testing this @danzimmerman !

traversaro, 2021-02-02:
Hi @danzimmerman , I recently tried to spawn your `gz-simple.yml` environment, and it seems to get the latest version of ogre 1.10.12 . I am not sure what the specific issue one month ago, but I guess there were a few migration going on conda-forge that were not complete for Gazebo (libprotobuf314 in particular) that could have affected the installation process.

danzimmerman, 2021-02-05:
Cool, will try it again soon.

danzimmerman, 2021-02-22:
Conda-installing gazebo (and ROS Noetic) are working for me now.

wolfv, 2021-02-22:
That's great to hear!

Y-Grigoriy, 2021-02-24:
> Conda-installing gazebo (and ROS Noetic) are working for me now.

Hello, thank you very much for your solution. Did you install ROS Noetic via Conda? And could you tell how the ROS can be connected with Gazebo? Sorry, if it is off-topic of this issue.

traversaro, 2021-02-24:
> Did you install ROS Noetic via Conda?

This is possible via the RoboStack channel, see the following link for reference:
* https://github.com/RoboStack/ros-noetic
* https://medium.com/robostack/cross-platform-conda-packages-for-ros-fa1974fd1de3

> And could you tell how the ROS can be connected with Gazebo? 

ROS can be connected with Gazebo via the `gazebo_ros_pkgs`, see the Gazebo/ROS documentation in http://gazebosim.org/tutorials?tut=ros_overview for more info.

In general, for this kind of question a Answers&Question website like https://answers.ros.org/questions/ or https://answers.gazebosim.org/questions/ .

FirefoxMetzger, 2021-02-25:
```
System: Win10 Build 19042.804
Graphics: Geforce RTX 3070, driver version 461.40, DirectX runtime 12.
Conda version: (min)conda 4.9.2, gazebo version 11.3.0
Other: None (antivirus, firewall, etc. enabled)
```

I had to run the terminal as administrator (probably because conda is installed for all users), otherwise no problems during the installation.

When I start gazebo, I get 2 new processes. `gzserver` (flagged as non-responding by the task manager, but appears to work fine), and `gzclient`. The latter spawns two windows, one called gzclient (as expected) and one called OgreWindow(0), which doesn't seem to do much, and closing it doesn't seem to have any obvious consequences.

Admitedly, I haven't tested running any simulation or loading models yet. Is there a shared test scenario/world I should try?

traversaro, 2021-02-25:
> When I start gazebo, I get 2 new processes. gzserver (flagged as non-responding by the task manager, but appears to work fine), and gzclient. The latter spawns two windows, one called gzclient (as [...]

The gazebo command also for Windows was added recently in https://github.com/osrf/gazebo/pull/2864, so there could be problems. However, indeed it works by spawning two separate gzserver and gzclient process. I am not sure instead about the `Ogre Window()` window, but I guess that is also spawned if you launch gzclient on its own.

ImanHosseini, 2021-04-04:
I use conda command prompt, and I get stuck on "[Msg] Waiting for master." weirdly, it randomly works sometimes [everything works] but most of the time it just gets stuck in "[Msg] Waiting for master." and it's not even generating any network traffic, and it's really weird, what I'm wondering is, why does it work? I don't get it, the few times that it did work, what was different?! Can this be a multi threading bug somewhere?

wolfv, 2021-04-04:
Can you send your OS, and the output of conda list?

wolfv, 2021-04-04:
Or also conda info

ImanHosseini, 2021-04-04:
win64.
conda info:
```
     active environment : r2
    active env location : C:\Users\salva\miniconda3\envs\r2
            shell level : 2
       user config file : C:\Users\salva\.condarc
 populated config files : C:\Users\salva\.condarc
          conda version : 4.9.2
    conda-build version : not installed
         python version : 3.8.5.final.0
       virtual packages : __cuda=11.2=0
                          __win=0=0
                          __archspec=1=x86_64
       base environment : C:\Users\salva\miniconda3  (writable)
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/robostack/win-64
                          https://conda.anaconda.org/robostack/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\salva\miniconda3\pkgs
                          C:\Users\salva\.conda\pkgs
                          C:\Users\salva\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\salva\miniconda3\envs
                          C:\Users\salva\.conda\envs
                          C:\Users\salva\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.9.2 requests/2.24.0 CPython/3.8.5 Windows/10 Windows/10.0.21296
          administrator : False
             netrc file : None
           offline mode : False
```
conda list:
```
# Name                    Version                   Build  Channel
apr                       1.7.0                he38c35c_5    conda-forge
assimp                    5.0.1                hc2aa0de_5    conda-forge
bcrypt                    3.2.0            py38h294d835_1    conda-forge
boost                     1.74.0           py38h1266d08_3    conda-forge
boost-cpp                 1.74.0               h54f0996_2    conda-forge
bullet                    3.09                 h57928b3_1    conda-forge
bullet-cpp                3.09                 h60cbd38_1    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2020.12.5            h5b45459_0    conda-forge
catkin_pkg                0.4.23             pyh9f0ad1d_0    conda-forge
certifi                   2020.12.5        py38haa244fe_1    conda-forge
cffi                      1.14.5           py38hd8c33c5_0    conda-forge
cmake                     3.19.7               h39d44d4_0    conda-forge
console_bridge            1.0.1                h5362a0b_0    conda-forge
cppzmq                    4.7.1                h4324990_2    conda-forge
cryptography              3.4.7            py38hd7da0ea_0    conda-forge
curl                      7.76.0               hf1763fc_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.5.0              pyh9f0ad1d_0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
docutils                  0.16             py38haa244fe_3    conda-forge
eigen                     3.3.9                h2d74725_1    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.3.0                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
freeglut                  3.2.1                h0e60522_2    conda-forge
freeimage                 3.18.0               hb49e22e_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
gazebo                    11.4.0               h83ca5bb_0    conda-forge
gettext                   0.19.8.1          h1a89ca6_1005    conda-forge
[... 203 lines cut ...]
sdl_image                 1.2.12               h368d6c2_1    conda-forge
setuptools                49.6.0           py38haa244fe_3    conda-forge
sip                       4.19.20          py38h6538335_0    conda-forge
six                       1.15.0             pyh9f0ad1d_0    conda-forge
sqlite                    3.35.3               h8ffe710_0    conda-forge
tbb                       2020.2               h2d74725_4    conda-forge
tbb-devel                 2020.2               h2d74725_4    conda-forge
tiny-process-library      2.0.4                h0e60522_0    conda-forge
tinyxml                   2.6.2                h2d74725_2    conda-forge
tinyxml2                  8.0.0                h39d44d4_1    conda-forge
tk                        8.6.10               h8ffe710_1    conda-forge
urdfdom                   2.3.3                h7ef1ec2_0    conda-forge
urdfdom_headers           1.0.5                h5362a0b_2    conda-forge
vc                        14.2                 hb210afc_4    conda-forge
vs2015_runtime            14.28.29325          h5e1d092_4    conda-forge
wheel                     0.36.2             pyhd3deb0d_0    conda-forge
wincertstore              0.2             py38haa244fe_1006    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                he774522_0    conda-forge
yaml-cpp                  0.6.3                ha925a31_4    conda-forge
zeromq                    4.3.4                h0e60522_0    conda-forge
zlib                      1.2.11            h62dcd97_1010    conda-forge
zstd                      1.4.9                h6255e5f_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```

wolfv, 2021-04-04:
Thanks for tracking the bug down! I wonder if this is a general gazebo-on-windows problem or specific to conda-forge?

ImanHosseini, 2021-04-04:
It seems windows specific. If we just document this behaviour somewhere, it would be fine, so people know to kill zombie gzclient/gzserver/gazebo process if they get stuck in "Waiting for master." and everything else is fine, and it doesn't happen that much, like, if you start gazebo, and prematurely kill it superfast before it's all up, or some unnatural thing like that this issue might happen. I tried investigating further but it gets even weirder => if you do this on purpose (it may take multiple time to get the behaviour) and then try to attach a debugger to see what is the state of that "zombie" process, it just exits right away [the new gazebo would be -unstuck- and start working] not letting you attach, if you try dumping it => you get a broken empty dump.

traversaro, 2021-04-06:
`gzserver` not properly closing and preventing new gzserver to be spawned tend to happen also on other platforms, there are a few PRs working on it: 
* https://github.com/osrf/gazebo/pull/2949
* https://github.com/osrf/gazebo/pull/2950

Perhaps @ivanpauno can provide tips on how debug further failures of this kind. 

> If we just document this behaviour somewhere, it would be fine, so people know to kill zombie gzclient/gzserver/gazebo process if they get stuck in "Waiting for master." and everything else is fine, [...]

This make sense. I think at this point given the amount of feedback on this issue, it could make sense if we move some of the knowledge contained in this issue (i.e. how to install it and a few known issues, such as the one that you mentioned) in a piece of docs in https://github.com/osrf/gazebo_tutorials , what do you think @j-rivero ?

j-rivero, 2021-04-06:
> This make sense. I think at this point given the amount of feedback on this issue, it could make sense if we move some of the knowledge contained in this issue (i.e. how to install it and a few [...]

I think that you guys did a great work on debugging and clarifying the whole thing, thanks so much. I would be happy to review a PR against gazebo_tutorials. If you don't want to deal with the format of the docs, I can translate a markdown format if that helps. Feel free to assign the PR to me. I'm looking for a bit of time to test the conda installation again.

ivanpauno, 2021-04-06:
> Perhaps @ivanpauno can provide tips on how debug further failures of this kind.

On windows?
Oh sorry, my debugging abilities on that platform are terrible :joy:.

In linux what I did was finding a simulation where I could reproduce the issue.
In my case I was spawning a robot in a gazebo world with some sensor plugins, like `DepthCameraPlugin`, and using a launch file to set up that.
I then started and ctrl-c the simulation repeatedly until I found `gzserver` hanging.
Then, attached with gdb, printed a multithreaded traceback a few times to see which threads were making progress, and started the guessing work to see what was blocking what.

I also added extra cpu stress to make the issue easier to reproduce.
For one of the bugs, I build gazebo from source with debugging symbols.

Doing something similar on Windows should be possible (e.g. https://docs.microsoft.com/en-us/visualstudio/debugger/attach-to-running-processes-with-the-visual-studio-debugger?view=vs-2019#:~:text=You%20can%20attach%20the%20Visual,the%20debugger%20to%20the%20process.), but I don't have experience in that.

traversaro, 2021-04-06:
> > Perhaps @ivanpauno can provide tips on how debug further failures of this kind.
> 
> On windows?
> Oh sorry, my debugging abilities on that platform are terrible 😂.

Don't worry, the step that you described apply to more and less any platform with a debugger, thanks! Furthermore, I actually suspect that most bug of this kind are rather multiplatform, but this is just my personal guess.

ImanHosseini, 2021-04-07:
> > Perhaps @ivanpauno can provide tips on how debug further failures of this kind.
> 
> On windows?
> Oh sorry, my debugging abilities on that platform are terrible 😂.
> 
> In linux what I did was finding a simulation where I could reproduce the issue.
> In my case I was spawning a robot in a gazebo world with some sensor plugins, like `DepthCameraPlugin`, and using a launch file to set up that.
> I then started and ctrl-c the simulation repeatedly until I found `gzserver` hanging.
> Then, attached with gdb, printed a multithreaded traceback a few times to see which threads were making progress, and started the guessing work to see what was blocking what.
> 
> I also added extra cpu stress to make the issue easier to reproduce.
> For one of the bugs, I build gazebo from source with debugging symbols.
> 
> Doing something similar on Windows should be possible (e.g. https://docs.microsoft.com/en-us/visualstudio/debugger/attach-to-running-processes-with-the-visual-studio-debugger?view=vs-2019#:~:text=You%20can%20attach%20the%20Visual,the%20debugger%20to%20the%20process.), but I don't have experience in that.

Thanks, this is really helpful! I did track down what's going on in my case (on windows) and it was indeed this issue: https://github.com/osrf/gazebo/pull/2950 and the suggested change seems to fix it.

traversaro, 2021-04-21:
> Thanks, this is really helpful! I did track down what's going on in my case (on windows) and it was indeed this issue: #2950 and the suggested change seems to fix it.

The change in  #2950  was included in Gazebo Classic 11.5.0, that was released yesterday and for which conda-forge binaries are now available: https://github.com/conda-forge/gazebo-feedstock/pull/67 . Hopefully then that problem was solved.

PrasadNR, 2021-08-29:
I am using Windows 8. I just did `conda install -c conda-forge gazebo` and then simply typed `gazebo` in the Anaconda prompt. This Gazebo does not work sometimes and sometimes it does (Same "Windows is finding a solution to the problem" error). I'm not sure why though.

traversaro, 2021-08-30:
> 
> 
> I am using Windows 8. I just did `conda install -c conda-forge gazebo` and then simply typed `gazebo` in the Anaconda prompt. This Gazebo does not work sometimes and sometimes it does (Same "Windows is finding a solution to the problem" error). I'm not sure why though.

Hi @PrasadNR, thanks for the report! Indeed I am not sure anyone ever tested this on Windows 8. Could you open an issue in https://github.com/conda-forge/gazebo-feedstock/issues/new, with all the information required by the issue template? In particular, the list of packages that you obtain with `conda list` are particular interesting. 

Other things that could be helpful for debug: 
* Run `gazebo --verbose` 
* Try to first run `gzserver --verbose` in a terminal and `gzclient --verbose` in another 

I wonder if this comment https://github.com/conda-forge/vc-feedstock/pull/37#issuecomment-861072860 is related. I remember that when I read it I tought it was a purely teoretical problem, but that is not the case it seems. : )