gazebosim/gazebo-classic #2900: Gazebo on Windows: superbuild-vcpkg

j-rivero, 2020-12-09:
The current documentation and CI for Gazebo works by using a set of custom tarballs created manually long ago. To replace this method, I thought on using vcpkg to generate a workspace with dependencies and inject that in the Gazebo build. It happens that the great @traversaro already has this ready and is generating binaries to install the whole thing: https://github.com/robotology/robotology-superbuild-dependencies-vcpkg. This method will allow people not only to run Gazebo but it is useful to develop software using it since the sources and dependencies are easily available.

**Help wanted:**
We are interested in knowing if our users on Windows can install and use it without problems. If you want to help, please follow 
[the instructions in the README](https://github.com/robotology/robotology-superbuild-dependencies-vcpkg/blob/master/README.md). To help with the testing, please comment in the issue with details about Windows version, superbuild version installed, problems/not-problems found, graphic card drivers installed and other information you find useful. You are also welcome to comment and help with other people's problem.

**Windows specific:**
 * Be careful with the firewall rules
 * Be careful with Antivirus actions interfering Gazebo run

j-rivero, 2020-12-09:
My testing of the superbuild:
 * **System**: Win10 Home 64bits
 * **Graphics**: Geforce MX150 + Intel (configured in nvidia control panel to run gzserver/gzclient on nvidia), driver version 398.36, DirectX runtime 12.
 * **Superbuild**: [version 0.5.0](https://github.com/robotology/robotology-superbuild-dependencies-vcpkg/releases/tag/v0.5.0) 
 * **Other**: Disabled antivirus.

**Runtime testing**:
```batch
C:\robotogy\scripts\setup-deps.bat
C:\robotogy\scripts\setup-gazebo.bat
C:\robotogy\gazebo\install\setup.bat
gzserver --verbose  (accept firewall rules)
```
In a different terminal
```batch
C:\robotogy\scripts\setup-deps.bat
C:\robotogy\scripts\setup-gazebo.bat
C:\robotogy\gazebo\install\setup.bat
gzclient --verbose  (accept firewall rules)
```
And Gazebo appears!

traversaro, 2020-12-09:
In theory only `C:\robotogy\scripts\setup-deps.bat` is necessary, as that calls all other scripts.

traversaro, 2020-12-14:
To clarify, my long term plan at the moment is that if the conda-forge installation works out is to eventually discontinue the maintenance of https://github.com/robotology/robotology-superbuild-dependencies-vcpkg and just support conda-forge, especially for complex dependencies such as Gazebo. https://github.com/robotology/robotology-superbuild-dependencies-vcpkg may still remain in place for vcpkg-only dependencies, as it still has some advantaged over conda-forge binaries (mainly the availability of Debug libraries), but I would really prefer to get rid of the custom scripts to build and install the colcol workspace of Gazebo and its own dependencies. 

Having said that, if anyone wants to take that scripts and maintain them somewhere else, it would perfectly fine for me.

traversaro, 2022-03-22:
I am actively trying to get user to stop using those binaries, so I think we can avoid mentioning them and close this issue, thanks!