gazebosim/gazebo-classic #2901: Gazebo on Windows (metaticket)

j-rivero, 2020-12-09:
The current situation of the documentation for Windows installation of Gazebo is broken. The changes to the code base of Gazebo are going through Windows CI and patches for Windows are welcome and processed. Our intention is to get rid of the current installation method based on downloading different manually created tarballs and use something more stable.

To solve the situation we are going to need the Gazebo user community. This ticket will act as a meta-ticket for different actions dedicated to evaluate the possible options we have to have Gazebo installed and running on Windows for both, users and developers. For specific details of every implementations please refer to the corresponding linked issues. Feel free to comment here with new proposals or comments that don't fit well into the different approaches.

Proposals:

 - [ ] Test conda-forge installation for Gazebo users: #2899
 - [ ] Test @traversaro [superbuild-vcpkg](https://github.com/robotology/robotology-superbuild-dependencies-vcpkg) for Gazebo development and use: #2900

ewillsbae, 2021-01-14:
Is there a ticket for the ms-iot dependencies? I've been working with those pretty heavily so far to try my hand at getting a stable ROS/Gazebo installation on Windows.

j-rivero, 2021-01-19:
> Is there a ticket for the ms-iot dependencies?

We don't have one right now but I'm happy to add it to this tracker if you create one in the same way we have them for vcpkg or conda. Thanks.

jemeeshat, 2021-02-02:
Hello All, 
I am a beginner in programming, I have no experience at all!
I am installing Gazebo following the manual for windows 10 (64). I am having issues in step 8 and I do not know how to resolve it

Would anyone help me, please?

traversaro, 2021-02-02:
Hi @jemeeshat, especially  if you are not an expert in programming, I think that the easiest procedure to test of the one mentioned in this issue is the `conda-forge` one tracked in https://github.com/osrf/gazebo/issues/2899 . 

There is no explicit documentation for it yet, but more and less it can be summarized as:
* Install and get familiar with a minimal conda distribution, such as [`Miniconda`](https://docs.conda.io/en/latest/miniconda.html) or [`Miniforge`](https://github.com/conda-forge/miniforge)
* Once you have conda installed, you can create a new environment with Gazebo as in the following:
~~~
conda create -n gazebo-env
conda activate robostackenv
conda install gazebo -c conda-forge
~~~

At this point, you should be able to launch gazebo . As you can find in https://github.com/osrf/gazebo/issues/2899 there are still open issue, but I think that overall the installation process is more doable if you do not have programming experience.

ankurash, 2021-03-03:
> Hi @jemeeshat, especially if you are not an expert in programming, I think that the easiest procedure to test of the one mentioned in this issue is the `conda-forge` one tracked in #2899 .
>
> There is [...]

conda activate gazebo-env

traversaro, 2021-03-03:
Thanks @ankurash , I fixed the original comment (copy&paste error!).

nkbeebe, 2021-06-11:
I receive an error on step 1 (the second step 1) with this command:

_cd ign-cmake
# if you want debug, run configure Debug
.\configure
nmake
nmake install_

I've attached the screenshot, but I'm also confused on the instructions of the step prior to this one:

_Open a regular Windows shell (Start->Run->"cmd"->enter), and load your compiler setup by copying and pasting the following line:

"C:\Program Files\Microsoft Visual Studio\2017\Community\VC\Auxiliary\Build\vcvarsall.bat" x86_amd64_

I have that .bat file, but its not in the same location ign-cmake is, and what is the x86_amd64 mean? I can simply open visual studio or visual studio code to that location, but from there what? And how does this relate to changing directory to configure ign-...

Thanks for the help

[image]
[image]

replicatesyndicate, 2021-07-18:
Hello,  
I'm particularly interested in installing `gazebo9`, as higher versions of Gazebo are not compatible with my particular application, and it also must be on Windows due to compatibility issues again.  
The alternative installers in #2899 and particularly #2900 seem to be really good, but they apparently install `gazebo11` which is not good in my case.  Is there a neat way of installing `gazebo9` in particular on a Windows machine?  
### If CMake is the only way to install `gazebo9`
If there's *not*, here's my progress with the old CMake method:
I've tried to tinker with the batch files `configure.bat` on each dependency, using the change below on *all* `configure.bat` files. Note the `DCMAKE_PREFIX_PATH` part.  
```
cmake .. -G "NMake Makefiles" -DCMAKE_INSTALL_PREFIX="%WORKSPACE_INSTALL_DIR%" -DCMAKE_BUILD_TYPE="%build_type%" -DBUILD_TESTING:BOOL=False -DCMAKE_PREFIX_PATH="<path_to_gz_ws>\deps"
```
This requires copying `\build\install\Release` installs, and the `C:\lib\cmake` folders into a single dependencies folder. I've matched their names with whatever element CMake claimed "missing". I've also located [`ign-tools`](https://github.com/ignitionrobotics/ign-tools) on GitHub as it is also required but not mentioned, and applied the same procedure.  
However, *some* other dependencies like ZIP and JSONCPP still break my CMake configurations as their, and I may need help with that.   
An example error message of such dependencies is below:
```
CMake Warning at C:/Gazebo/_deps/ignition-cmake0/cmake0/IgnPkgConfig.cmake:48 (message):
  The package [ZIP] requires pkg-config in order to be found.  Please install
  pkg-config so we can search for that package.
Call Stack (most recent call first):
  C:/Gazebo/_deps/ignition-cmake0/cmake0/FindZIP.cmake:30 (ign_pkg_check_modules)
  C:/Gazebo/_deps/ignition-cmake0/cmake0/IgnUtils.cmake:146 (find_package)
  CMakeLists.txt:43 (ign_find_package)
```
...
```
-- BUILD ERRORS: These must be resolved before compiling.
--      Missing: JSONCPP
--      Missing: ZIP
--  END BUILD ERRORS

CMake Error at C:/Gazebo/_deps/ignition-cmake0/cmake0/IgnConfigureBuild.cmake:64 (message):
  Errors encountered in build.  Please see BUILD ERRORS above.
Call Stack (most recent call first):
  CMakeLists.txt:60 (ign_configure_build)
```
I may require something called `pkg-config` for these, and renaming them appropriately so that the folder `pkgconfig` is reachable didn't help at all. I may even progress with these using that, but I feel like I'm reaching the limit of fixing things without breaking others.  
Thanks in advance.

traversaro, 2021-07-19:
>  Is there a neat way of installing `gazebo9` in particular on a Windows machine? Would giving WSL2 a try help?

If for your specific needs WSL/WSL2 is a possible option, for sure installing Gazebo 9 will be much easier. Unless you have the latest WSLg ( https://github.com/microsoft/wslg ) that is however not available in stable Windows builds, you just need to setup a X server on the Windows side (a few instructions on this can be found in https://github.com/robotology/robotology-superbuild#windows-subsystem-for-linux-from-source), and then you should be able to install Gazebo 9 as you install it in Ubuntu via apt. However to avoid the pitfalls of https://en.wikipedia.org/wiki/XY_problem it may be easier to give you the right suggestions if you explain a bit more your use case.

mtnrbq, 2021-07-27:
> > Is there a neat way of installing `gazebo9` in particular on a Windows machine? Would giving WSL2 a try help?
> 
> If for your specific needs WSL/WSL2 is a possible option, for sure installing Gazebo 9 will be much easier. Unless you have the latest WSLg ( https://github.com/microsoft/wslg ) that is however not available in stable Windows builds, you just need to setup a X server on the Windows side (a few instructions on this can be found in https://github.com/robotology/robotology-superbuild#windows-subsystem-for-linux-from-source), and then you should be able to install Gazebo 9 as you install it in Ubuntu via apt. However to avoid the pitfalls of https://en.wikipedia.org/wiki/XY_problem it may be easier to give you the right suggestions if you explain a bit more your use case, as the "compatibility issues" that required you to be on Windows may also prevent you to use WSL2, depending on the nature of this issues.

In reference to this, I'm running wslg, which is working well with a couple of other apps - gtkwave and quartus II. gazebo however does not work either as-is or with a separate xserver - would be interested if it 'should just work' on wslg.

traversaro, 2021-07-27:
> In reference to this, I'm running wslg, which is working well with a couple of other apps - gtkwave and quartus II. gazebo however does not work either as-is or with a separate xserver - would be interested if it 'should just work' on wslg.

That is interesting. I do not have a machine with wslg to test, but Gazebo was one of the application used by Microsoft itself to showcase the wslg capabilities, see https://devblogs.microsoft.com/commandline/the-initial-preview-of-gui-app-support-is-now-available-for-the-windows-subsystem-for-linux-2/#bonus-leverage-wsls-gpu-access-to-run-linux-applications-with-3d-acceleration . Which problem are you experiencing in particul such that "gazebo however does not work" ? Perhaps it could be worth to report the issue in https://github.com/microsoft/wslg . I found https://github.com/microsoft/wslg/issues/185, but I do not know if it is related.

mtnrbq, 2021-07-27:
> > In reference to this, I'm running wslg, which is working well with a couple of other apps - gtkwave and quartus II. gazebo however does not work either as-is or with a separate xserver - would be interested if it 'should just work' on wslg.
> 
> That is interesting. I do not have a machine with wslg to test, but Gazebo was one of the application used by Microsoft itself to showcase the wslg capabilities, see https://devblogs.microsoft.com/commandline/the-initial-preview-of-gui-app-support-is-now-available-for-the-windows-subsystem-for-linux-2/#bonus-leverage-wsls-gpu-access-to-run-linux-applications-with-3d-acceleration . Which problem are you experiencing in particul such that "gazebo however does not work" ? Perhaps it could be worth to report the issue in https://github.com/microsoft/wslg . I found [microsoft/wslg#185](https://github.com/microsoft/wslg/issues/185), but I do not know if it is related.

Hmm, very interesting - thanks for the link, I had not seen that one.
I've gone down the path of a conda install, and have just kicked off gazebo, but not 100% sure of how long I should expect 'preparing your world' to take (over 5 minutes so far)

traversaro, 2021-07-27:
>  I've gone down the path of a conda install, and have just kicked off gazebo, but not 100% sure of how long I should expect 'preparing your world' to take (over 5 minutes so far)

Did you tried to install gazebo via conda on Linux on WSL and then you are trying to run it via wslg or did you installed gazebo via conda on Windows directly?

mtnrbq, 2021-07-27:
> Did you tried to install gazebo via conda on Linux on WSL and then you are trying to run it via wslg or did you installed gazebo via conda on Windows directly?

Gazebo via conda on Windows

traversaro, 2021-07-27:
> Gazebo via conda on Windows

Ack, for that could you open a separate issue in https://github.com/conda-forge/gazebo-feedstock, so it is easier to track this? Thanks!

replicatesyndicate, 2021-07-27:
My particular question of course wouldn't involve conda as it is exclusively used for the latest version of Gazebo. WSLg seems to be the best workaround, so I'm just waiting for updates.

> However to avoid the pitfalls of https://en.wikipedia.org/wiki/XY_problem it may be easier to give you the right suggestions if you explain a bit more your use case, as the "compatibility issues" [...]

The specific use case for me is due to the co-simulation requirements of Simulink:  
[Perform Co-Simulation between Simulink and Gazebo](https://www.mathworks.com/help/robotics/ug/perform-co-simulation-between-simulink-and-gazebo.html)  
The OS requirement suggests Bionic, but Focal also works. The simulation on the Simulink end must run on Windows due to other dependencies not being available on Linux. (This is particularly why the WSL option is better)
> Operating System: Ubuntu Xenial Xerus or Ubuntu Bionic Beaver
Software dependency: CMake 2.8, Gazebo 9, and libgazebo9-dev

I don't specifically need to run them simultaneously, as I occasionally have access to another computer to run the co-simulation. I need Gazebo on my computer to modify models and make individual test runs. However, I cannot simply get Ubuntu installed on my computer to run Gazebo due to some hardware issues (neither on a VM nor on dual-boot). So the best option for me would be getting Gazebo 9 to run on Windows directly.   

Sorry for taking long to reply, and thanks!

traversaro, 2021-07-27:
Cool, this is much more clear! In this case, I think that Co-Simulation with Simulink is always happening over network interfaces, so using WSL should work fine. Even without waiting for WSLg, using an external X server should be working fine.

traversaro, 2021-07-27:
> The specific use case for me is due to the co-simulation requirements of Simulink:
> [Perform Co-Simulation between Simulink and Gazebo](https://www.mathworks.com/help/robotics/ug/perform-co-simulation-between-simulink-and-gazebo.html)

By the way, I quickly inspected the CMake and C++ source code for the Gazebo plugin for co-simulation installed by the `packageGazeboPlugin`, at it seems that as of MATLAB R2021a, it is quite Linux-specific, so unless this is fixed in future MATLAB versions  or someone else fixes it, running it under WSL it is not only the most convenient option, but the only possible one, as such plugin will never be able to compile and link against a native Windows Gazebo such as the one installed via conda on Windows, Chocolatey or vcpkg.

replicatesyndicate, 2021-07-27:
Good to know, thanks! It'd never hurt to actually run the entire thing on a single computer.

tayloryoung6396, 2021-09-13:
Has anyone actually had luck installing on Windows 11 with wslg? When I go to run it
```
```taylor@DESKTOP-A9RLPRC:/mnt/c/Users/Taylor/Downloads$ gazebo
NVD3D10: CPU cyclestats are disabled on client virtualization
NVD3D10: CPU cyclestats are disabled on client virtualization
NVD3D10: CPU cyclestats are disabled on client virtualization
NVD3D10: CPU cyclestats are disabled on client virtualization 
taylor@DESKTOP-A9RLPRC:/mnt/c/Users/Taylor/Downloads$`
```

Ace314159, 2022-02-17:
I have created https://github.com/microsoft/vcpkg/pull/22466, which will add gazebo11 to vcpkg. Once it's merged (or by currently using my fork), you can install gazebo by using the command `vcpkg install --triplet=x64-windows gazebo` and opening the `gazebo` executable in `installed/x64-windows/tools/gazebo`