gazebosim/gazebo-classic #2908: Gazebo Crashes from velodyne 3D Lidar sensor

gauravbhosale, 2020-12-17:
Hi,

When I run attached world file as 'gazebo velodyneSensorTest.world --verbose', Gazebo 9.16 waits for some time and crashes after few seconds.
Is there anything wrong in .world file?

<?xml version="1.0"?>
<sdf version="1.4">
  <world name="default">
    <include>
      <uri>model://ground_plane</uri>
    </include>

    <include>
      <uri>model://sun</uri>
    </include>

    <model name="velodyne">
      <pose>-3 0 0.1 0 0 0</pose>
      <link name="link">
        <inertial>
          <mass>0.1</mass>
        </inertial>

        <collision name="collision">
          <geometry>
            <box>
              <size>0.2 0.2 0.2</size>
            </box>
          </geometry>
        </collision>

        <visual name="visual">
          <geometry>
            <box>
              <size>0.2 0.2 0.2</size>
            </box>
          </geometry>
        </visual>

        <sensor name="laser" type="ray">
          <pose>0.01 0 0.0175 0 -0 0</pose>
          <ray>
            <scan>
              <horizontal>
                <samples>640</samples>
                <resolution>1</resolution>
                <min_angle>-2.26889</min_angle>
                <max_angle>2.268899</max_angle>
              </horizontal>
              <vertical>
                <samples>32</samples>
                <resolution>1</resolution>
                <min_angle>-0.535</min_angle>
                <max_angle>0.186132</max_angle>
              </vertical>
            </scan>
	  <noise>
            <type>gaussian</type>
            <mean>0.0</mean>
            <stddev>0.01</stddev>
          </noise>
            <range>
              <min>0.1</min>
              <max>10</max>
              <resolution>0.01</resolution>
            </range>
          </ray>

          <always_on>1</always_on>
          <update_rate>200</update_rate>
          <visualize>true</visualize>
        </sensor>
      </link>
    </model>
</world>
</sdf>

[velodyneSensorTest.txt](https://github.com/osrf/gazebo/files/5709038/velodyneSensorTest.txt)

chapulina, 2020-12-17:
I don't see anything suspicious. Do you see any error messages if you run with `--verbose`?

gelardrc, 2021-01-04:
i think is just the <noise>, its not correctly indent.

Aladdin-Ark, 2021-01-04:
I can not test this world right now but from what I see this is not 360 coverage if this what you mean by velodyne 3D.
Also the update rate is kind of high, not to mention it is not gpy_ray. Which also make it suspicious with the collision you are using in the model.
I can not say why it would crash but it would be better to have it right to make sure it is not your model issue!
Try to reduce the update rate, and reduce the size of the collision to something like  <size>0.01 0.01 0.01</size>...

gauravbhosale, 2021-01-04:
Hi,

I am able to resolve this issue.
To resolve this issue,
1] I need to update Gazebo to latest version
2] I need to change VM setting and increase RAM to 4 GB ( previously it was 2 GB)

This means, when I use Velodyne sensor, the VM should have 4GB RAM.
Is this make sense? Is Gazebo needs 4 GB RAM always?

WCjai, 2021-01-04:
comparing <horizontal> ray and <vertical> ray , vertical ray reduces fps than horizontal ray... i don't know why.