gazebosim/gazebo-classic #2913: Depth camera corrupting GPU ray sensor data when clouds are enabled

pritzvac, 2021-01-07:
Initially I commented about this on issue #2500 but it looks like these are two separate problems. Furthermore, the problem persists in Gazebo 11.

When a world contains a GPU ray based sensor and a depth camera and the clouds are enabled, the GPU ray sensor data are corrupted each time the depth camera renders.

## How to replicate
Download the attached ROS package: [depth_camera_bug.zip](https://github.com/osrf/gazebo/files/5781549/depth_camera_bug.zip), copy it to a ROS workspace, build and run
`roslaunch depth_camera_bug depth_camera_bug.launch`.
Visualize the output of the `/laser_scan` topic in Rviz. You can use the *rviz.config* file in the provided package. Now the scan should be correct as in the first image.
Subscribe to the `/depth_cam/image_raw` topic of the depth camera either in Rviz or by running `rostopic echo /depth_cam/image_raw`. The laser scan will become corrupted as can be seen on the second image. (circles with their center at the depth camera position and with the radius of maximal range of the depth camera appear in the GPU ray sensor data)
[image]
[image]

The  package contains a world containing a sensor based on *libgazebo_ros_gpu_laser.so* and a depth camera sensor based on *libgazebo_ros_depth_camera.so*.
When the clouds in the world are enabled using the following tags
```
<sky>
  <clouds/>
</sky>
```
and both the GPU ray sensor and depth camera sensor are publishing data, the GPU ray sensor data will be corrupted.

## Possible solution
I managed to fix the problem by commenting out the call to `Camera::RenderImpl()` in the `DepthCamera::RenderImpl()` in *DepthCamera.cc*:
https://github.com/osrf/gazebo/blob/63d11b695134c7075161f1bdc36b91db47f27f92/gazebo/rendering/DepthCamera.cc#L620

But I guess this is not the correct solution to this problem and there is some other underlying issue, which is causing it, as the problem only appears when the clouds are turned on?

## Note about issue #2500
I also managed to replicate the issue #2500 in the same world when using a GPU ray sensor based on the Velodyne plugin instead of the depth camera. The plugin I used is this: https://github.com/ctu-mrs/mrs_gazebo_common_resources/blob/master/src/sensor_and_model_plugins/3dlidar_plugin.cpp
In this case, the scan from the Velodyne-based plugin is partially corrupted until the other GPU ray sensor renders for the first time (i.e. until something subscribes to its data and triggers the render function). This behavior seems to be fixed when I change the line
https://github.com/osrf/gazebo/blob/63d11b695134c7075161f1bdc36b91db47f27f92/gazebo/rendering/GpuLaser.cc#L755
to 
```C++
this->dataPtr->visual->SetVisible(false);
```
The visual visibility is set to `false` after each render in the `GpuLaser::RenderImpl()` function anyway, which is the reason why the data are not corrupted after the first render of the GPU ray sensor. However, this problem is probably not related to the depth camera bug.

iche033, 2021-01-07:
I remember running into this issue and we disabled cloud rendering in gpu lidar sensors on the server side:
https://github.com/osrf/gazebo/blob/gazebo11/gazebo/sensors/GpuRaySensor.cc#L340

It sounds like that line no longer works. Do you see clouds in in a typical rgb camera sensor (not gui camera)?

pritzvac, 2021-01-08:
I don't see the clouds but I still see the sky. Maybe the problem is in something else related to SkyX. I don't have the problem when I disable the sky completely (by deleting the `<sky></sky>` tags in the .world file).