gazebosim/gazebo-classic #2915: Unable to use gui plugins with gzclient

Jaeyoung-Lim, 2021-01-16:
I am trying to use a system plugin with gzclient. 

However, the `--gui-client-plugin` doesn't seem to work as expected as explained in http://gazebosim.org/tutorials?tut=system_plugin&cat=write_plugin

When I run gzclient with the following command
```
gzclient --verbose --gui-client-plugin libgazebo_user_camera_plugin.so
```

I get the following error: 
```
$ [Err] [RenderWidget.cc:314] System is attempting to load a plugin, but detected an incorrect plugin type. Plugin filename[libgazebo_user_camera_plugin.so].
```

I have copy pasted the exact system plugin example from https://github.com/osrf/gazebo/blob/gazebo9/examples/plugins/system_gui_plugin/system_gui.cc

**Additional Context**
I am running Ubuntu 20.04 with gazebo 11.3.0

chapulina, 2021-01-28:
I believe `--gui-client-plugin` is meant to load `GuiPlugin`s, while `-g` is meant to load `SystemPlugin`s into the gzclient process. The notes on the migration guide are a bit confusing. Here's the original PR for reference: https://osrf-migration.github.io/gazebo-gh-pages/#!/osrf/gazebo/pull-requests/2716/page/1

Jaeyoung-Lim, 2021-02-03:
@chapulina Thanks for the clarification. That is a bit clearer now.

closed 2021-02-03 by Jaeyoung-Lim