gazebosim/gazebo-classic #2916: can not use moveit to control robot with dart engine

zhongxiangzju, 2021-01-20:
The configuration is :
ros melodic
gazebo9 install from source following [http://gazebosim.org/tutorials?tut=install_from_source&cat=install](url)
dart6
moveit

when i set physics engine to **ode**, the robot can be moved using moveit. 
**But not the same if using dart.**

chapulina, 2021-02-02:
What happens when you use DART? Can you also tell what's the exact Gazebo 9 version you're using? Also, please try running in verbose mode to see what errors show up.

zhongxiangzju, 2021-07-14:
- I use Ubuntu 18.04 with ROS Melodic and Gazebo9.14.
- When using ODE physics engine, robot can be controlled using moveit.
- When change physics engine from ODE to DART, errors appears when launching moveit:
[ERROR] [1626272243.837366455]: Tried to advertise a service that is already advertised in this node
[ERROR] [1626272260.324671117, 14.718000000]: Ignoring transform for child_frame_id "forearm_link" from authority "unknown_publisher" because of a nan value in the transform (nan nan nan) (nan nan nan nan)

Then moveit shutdown and robot in gazebo is 
[image]

Screen shot of errors:
[image]

xiaoaxing, 2021-12-26:
Have you solved this problem, i meet the same problem.