gazebosim/gazebo-classic #2920: Marker example not behaving as expected due to id 0

rfn123, 2021-01-21:
Hi all,

I have looked into the [marker example ](https://github.com/osrf/gazebo/blob/gazebo11/examples/stand_alone/marker/marker.cc#L47) and  according to the description in line 47, the commands from line 47-63 should _modify_ the existing marker. However in reality, new markers are created. 

Reason is the id 0 that is being set for the first marker: [this line](https://github.com/osrf/gazebo/blob/67c43463c87685b7c9d485a9f8d2e6ac760815b9/gazebo/rendering/MarkerManager.cc#L214) indicates that sending id 0 will result in a randomized unique id. 
Probably the example should be changed to using another id.

chapulina, 2021-02-02:
Thanks for the ticket. I agree that the example should be updated.

DhruvMaroo, 2021-05-28:
Hey there! Thanks for bringing up this issue @rfn123. I have fixed it in my PR (#3014). Along with changing the set_id functions arguments (to make them non-zero for all the markers), I also have updated the CMakeLists.txt to use the newer versions of Ignition's transport and msgs packages. @chapulina Could you  please have a look at the PR and merge it? Thank you.

closed 2022-03-17 by rfn123