gazebosim/gazebo-classic #2928: WideAngleCamera cube map texture pixel format should be customizable

scpeters, 2021-01-27:
The `rendering::WideAngleCamera::CreateEnvRenderTexture` function creates an `envCubeMapTexture` using the [same pixel format](https://github.com/osrf/gazebo/blob/gazebo11_11.3.0/gazebo/rendering/WideAngleCamera.cc#L663) as the output image. For advanced users who write their own shaders, it can be necessary to use a floating point pixel format for the cube map texture even if the output image uses integers, for example to avoid integer overflow. I think it would be convenient to specify this in SDFormat as a `env_texture_format` parameter that is a sibling to the [//camera/lens/env_texture_size](https://github.com/osrf/sdformat/blob/sdformat10_10.2.0/sdf/1.6/camera.sdf#L133-L135) element.

scpeters, 2021-01-28:
Intending to target this at gazebo9 and gazebo11

iche033, 2021-01-28:
the proposed approach should work. One thing to note is that with `//camera/lens/env_texture_size` we're getting into implementation details on internal / intermediate texture formats. I'm not sure if the param would be useful other consumers of sdformat since they may decide to implement wide angle cameras differently. 

Other options include:
* setting the format to `Ogre::PF_FLOAT32_RGB` [here](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/rendering/WideAngleCamera.cc#L663) in gazebo classic and wide angle cameras should still work - it just uses more memory.
* Provide a `WideAngleCameraClass::SetEnvTexFormat` function for users to set through plugins. The downside of this approach is that we'll need to recreate the env textures when users sets a different format.

scpeters, 2021-01-30:
> * setting the format to `Ogre::PF_FLOAT32_RGB` [here](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/rendering/WideAngleCamera.cc#L663) in gazebo classic and wide angle cameras should still work - it just uses more memory.

I'll try this and see if I can quantify the memory impacts

scpeters, 2021-01-30:
> > * setting the format to `Ogre::PF_FLOAT32_RGB` [here](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/rendering/WideAngleCamera.cc#L663) in gazebo classic and wide angle cameras should still work - it just uses more memory.
> 
> I'll try this and see if I can quantify the memory impacts

I tried this earlier today with a pair of wide-angle cameras with 2048x2048 resolution with a target update rate of 30 Hz. When using the same intermediate pixel format, I was able to reach 30 frames per second easily, but when using `FLOAT32` for the cube map texture, it dropped the frame rate by a factor of 10 to less than 3 frames per second. According to `nvidia-smi`, it also increased the GPU memory usage of gzserver by a little over 1 GB, but I think the performance hit is the most significant factor.

This will need to be configurable.

scpeters, 2021-02-01:
I'll try adding an API for plugins as well as a custom namespaced element like `//camera/lens/gazebo:env_texture_format`

scpeters, 2021-02-04:
I've created a branch that sets the texture format based on the custom element `//camera/lens/gazebo:env_texture_format` in https://github.com/scpeters/gazebo/commit/d3793e344891d6a3224ab9b42b6ad8b934a217da

In terms of adding an API for setting the texture format programmatically, is it possible to change the texture after the camera has been initialized? This seems a bit more complicated to me

iche033, 2021-02-04:
I think you'll need to [remove](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/rendering/WideAngleCamera.cc#L499) the existing texture and then [create](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/rendering/WideAngleCamera.cc#L499) a new one with the new format

PR #2960 (2021-04-13): Specify wide angle camera cube map texture format
This PR sets the texture format based on the custom element `//camera/lens/gazebo:env_texture_format`.

Fixes #2928.
closed 2021-04-13 by scpeters