gazebosim/gazebo-classic #2930: [Err] [DARTPhysics.cc:771] The use of the ODE collision detector with DART is disabled because it causes conflicts with the version of ODE used in Gazebo.

trns1997, 2021-02-02:
Hi,
My world file looks like the following super simple nothing fancy, I just want to user the dart physics engine with ode collision detector. But I get the following error: `[Err] [DARTPhysics.cc:771] The use of the ODE collision detector with DART is disabled because it causes conflicts with the version of ODE used in Gazebo.`
```
<?xml version="1.0"?>
<sdf version="1.4">
  <world name="default">
    <include>
      <uri>model://ground_plane</uri>
    </include>

    <include>
      <uri>model://sun</uri>
    </include>

    <include>
      <uri>model://simple_box</uri>
    </include>

    <physics type="dart">
      <max_step_size>0.0005</max_step_size>
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>2000</real_time_update_rate>
      <collision_detector>ode</collision_detector>
    </physics>

  </world>
</sdf>
```

Anyone has any idea how to solve this problem? I build gazebo from source and followed the instructions.

scpeters, 2021-02-02:
have you tried using `bullet` for the `collision_detector`?

trns1997, 2021-02-03:
Yes I have, and the results are not the best plus it slows down the simulation. I also tried `dart` collision but that only seems to work with `boxshapes` and `spheres`. The moment you use complex meshes `dart` collision breaks. Are there any plans to get ode collision working with dart?