gazebosim/gazebo-classic #2931: Import ignition modules in python 2.7

NishanthARao, 2021-02-03:
Hi all.

I am trying to import ignition modules like transport and msgs on python2.7 .
I have installed ```libignition-transport4-dev``` and ```ibingition-msgs-dev```. But when I do `import ignition` i get an import error saying `No module named ignition`. Can someone please help me with this? Thank you

trns1997, 2021-02-03:
I am not sure it is as simple as that. Check out the following:  https://github.com/robotology/gym-ignition 
Not sure if this is what you want. But hopefully it is.

NishanthARao, 2021-02-03:
@trns1997 Thanks for your reply. Unfortunately, I too stumbled upon this earlier, and it is not what I want. Basically I want to plot visual markers on Gazebo like they did [here](https://www.youtube.com/watch?v=egWAMbU7f6Y) from python script directly. I guess that's not possible now?

trns1997, 2021-02-03:
The only way i see this possible is by writing your own gazebo plugin using C++, that spawns markers in the world. Other than that I am not sure how you could do this using python unless you find a python wrapper for gazebo. Sorry I can't be of more help.

NishanthARao, 2021-02-05:
@trns1997 Yeah, currently I think that's the only way to do it. Anyways, thank you for the suggestions.

chapulina, 2021-03-31:
Gazebo only provides C++ APIs. There are external projects like gym-ignition that provide python bindings to Ignition libraries, so I recommend you look into those.

closed 2021-03-31 by chapulina