gazebosim/gazebo-classic #2932: ign-msgs compile error

beijihuohu1, 2021-02-05:
The IDE is vs2019 x64.
     Protobuff lib is protobuf-3.4.1-vc15-x64-dll-MD.
     When i configure the ign-msgs with cmake,it's succssful,but when i build the project ignition-msgs1,it's fail,the output as follows:
1>Running protoc on E:/product/gs_ws/ign-msgs/proto/ignition/msgs/altimeter.proto
1>--ignmsgs_out: protoc-gen-ignmsgs: Plugin failed with status code 3221225781.
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(238,5): error MSB8066:
I dont known why.

j-rivero, 2021-02-05:
I'm afraid that we are going to need more parts of the log to know what's going on. I suggest that you look for `error C` or `error L` to see the exact problem raised by the compiler.

FirefoxMetzger, 2021-02-25:
I just ran into the same issue. Here is the full log

<details>
<summary>logs</summary>

```
gz-ws\ign-msgs\build>nmake

Microsoft (R) Program Maintenance Utility Version 14.28.29337.0
Copyright (C) Microsoft Corporation.  All rights reserved.

        "C:\Program Files\CMake\bin\cmake.exe" -S"C:\Users\Sebastian\Documents\Coding Projects\gz-ws\ign-msgs" -B"C:\Users\Sebastian\Documents\Coding Projects\gz-ws\ign-msgs\build" --check-build-system CMakeFiles\Makefile.cmake 0
        "C:\Program Files\CMake\bin\cmake.exe" -E cmake_progress_start "C:\Users\Sebastian\Documents\Coding Projects\gz-ws\ign-msgs\build\CMakeFiles" "C:\Users\Sebastian\Documents\Coding Projects\gz-ws\ign-msgs\build\\CMakeFiles\progress.marks"
        "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\bin\HostX86\x64\nmake.exe"  -f CMakeFiles\Makefile2 /nologo -                   all
        "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\bin\HostX86\x64\nmake.exe"  -f src\CMakeFiles\ign_msgs_gen.dir\build.make /nologo -L                  src\CMakeFiles\ign_msgs_gen.dir\depend
        "C:\Program Files\CMake\bin\cmake.exe" -E cmake_depends "NMake Makefiles" "C:\Users\Sebastian\Documents\Coding Projects\gz-ws\ign-msgs" "C:\Users\Sebastian\Documents\Coding Projects\gz-ws\ign-msgs\src" "C:\Users\Sebastian\Documents\Coding Projects\gz-ws\ign-msgs\build" "C:\Users\Sebastian\Documents\Coding Projects\gz-ws\ign-msgs\build\src" "C:\Users\Sebastian\Documents\Coding Projects\gz-ws\ign-msgs\build\src\CMakeFiles\ign_msgs_gen.dir\DependInfo.cmake" --color=
Dependencies file "src/CMakeFiles/ign_msgs_gen.dir/Generator.cc.obj.d" is newer than depends file "C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/build/src/CMakeFiles/ign_msgs_gen.dir/compiler_depend.internal".
Dependencies file "src/CMakeFiles/ign_msgs_gen.dir/generator_main.cc.obj.d" is newer than depends file "C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/build/src/CMakeFiles/ign_msgs_gen.dir/compiler_depend.internal".
Consolidate compiler generated dependencies of target ign_msgs_gen
        "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\bin\HostX86\x64\nmake.exe"  -f src\CMakeFiles\ign_msgs_gen.dir\build.make /nologo -L                  src\CMakeFiles\ign_msgs_gen.dir\build
[  1%] Built target ign_msgs_gen
        "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\bin\HostX86\x64\nmake.exe"  -f src\CMakeFiles\ignition-msgs1.dir\build.make /nologo -L                  src\CMakeFiles\ignition-msgs1.dir\depend
[  1%] Running protoc on C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/proto/ignition/msgs/altimeter.proto
        cd C:\Users\SEBAST~1\DOCUME~1\CODING~1\gz-ws\ign-msgs\build\src
        echo >nul && "C:\Users\Sebastian\Documents\Coding Projects\gz-ws\protobuf-3.15.2\cmake\build\Release\bin\protoc.exe" "--proto_path=C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/proto" "--plugin=protoc-gen-ignmsgs=C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/build/src/ign_msgs_gen.exe" "--cpp_out=dllexport_decl=IGNITION_MSGS_VISIBLE:C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/build/include" --ignmsgs_out "C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/build/include" "C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/proto/ignition/msgs/altimeter.proto"
--ignmsgs_out: protoc-gen-ignmsgs: Plugin failed with status code 3221225781.
NMAKE : fatal error U1077: 'echo' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\bin\HostX86\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\bin\HostX86\x64\nmake.exe"' : return code '0x2'
Stop.
```
</details>

traversaro, 2021-02-25:
`3221225781` is quite cryptic, but it actually means that the command: 
~~~
C:\Users\Sebastian\Documents\Coding Projects\gz-ws\protobuf-3.15.2\cmake\build\Release\bin\protoc.exe" "--proto_path=C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/proto" "--plugin=protoc-gen-ignmsgs=C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/build/src/ign_msgs_gen.exe" "--cpp_out=dllexport_decl=IGNITION_MSGS_VISIBLE:C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/build/include" --ignmsgs_out "C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/build/include" "C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/proto/ignition/msgs/altimeter.proto"
~~~

is failing due to some missing dll. I suggest to make sure that all the dll required by both `C:\Users\Sebastian\Documents\Coding Projects\gz-ws\protobuf-3.15.2\cmake\build\Release\bin\protoc.exe` and `C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/build/src/ign_msgs_gen.exe` are either in the directory that contains the .exe, or in the PATH. This can be easily verified by opening the .exe in the same terminal you are using to compile (to ensure that the PATH is the same) with https://github.com/lucasg/Dependencies . In particular, as you compile protobuf on your own, are you sure that all the dll of protobuf required by C:/Users/Sebastian/Documents/Coding Projects/gz-ws/ign-msgs/build/src/ign_msgs_gen.exe are either part of the `ign_msgs_gen.exe`'s directory or in the PATH?

traversaro, 2021-02-25:
Just as curiosity, there is any specific reason why you compiled protobuf on your own instead of relying on a package manager like vcpkg or conda?

FirefoxMetzger, 2021-02-25:
Solved 👍  

Protobuf's `/bin` folder  was not on my path, and adding it solved the problem. The error message (`-ignmsgs_out: protoc-gen-ignmsgs: Plugin failed with status code 3221225781.`) is quite cryptic...