gazebosim/gazebo-classic #2934: [Err] [ModelDatabase.cc:390] Unable to parse model.config for model[http://gazebosim.org/models/bin_4_dropping_task]

Maddlovski, 2021-02-14:
Hey guys, 
I have been trying to load a DEM in gazebo just like it is shown on the Gazebo DEM tutorial. (http://gazebosim.org/tutorials) 
Everytime I start Gazebo in the terminal it won't even show an error. It only shows the following error when I use --verbose in the command line: [Err] [ModelDatabase.cc:390] Unable to parse model.config for model[http://gazebosim.org/models/bin_4_dropping_task]

The URL does not exist when I try to open the link on the Internet.

Gazebo does actually open and it says the model is loaded but there is no model to find!
Does anyone know how to fix the Problem ?

Thanks in Advance, 
Martin

Lovemyse1f, 2021-02-26:
1. have you put the relative models into .gazebo/models? 
2. The model loaded is in the path/file?

Maddlovski, 2021-03-01:
Hello, thanks for your comment but I have already tried that without success!

pit36, 2021-03-12:
Hello, i got the same issue:

> [gazebo-1] [Err] [Node.cc:105] No namespace found
[gazebo-1] [Err] [ModelDatabase.cc:390] Unable to parse model.config for model[http://gazebosim.org/models/bin_4_dropping_task]
[gazebo-1] [Err] [Node.cc:105] No namespace found
[gazebo-1] [Err] [REST.cc:205] Error in REST request
[gazebo-1] 
[gazebo-1] libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'
[gazebo-1] [Err] [Node.cc:105] No namespace found
[gazebo-1] [Err] [Node.cc:105] No namespace found
[gazebo-1] [Err] [ModelDatabase.cc:390] Unable to parse model.config for model[http://gazebosim.org/models/bin_4_dropping_task]
[gazebo-1] [Err] [Node.cc:105] No namespace found
[gazebo-1] [Err] [Node.cc:105] No namespace found
[gazebo-1] [Err] [Node.cc:105] No namespace found
[gazebo-1] [Err] [Node.cc:105] No namespace found
[gazebo-1] [Err] [ModelDatabase.cc:414] Unable to download model[model://kmr]
[gazebo-1] [Err] [SystemPaths.cc:412] File or path does not exist[""]
[gazebo-1] Error Code: 11 Msg: Unable to find uri[model://kmr]

alextac98, 2021-03-28:
Hello, I also have this issue. I am not using any sdf models. It happens when I launch a URDF model with Gazebo, no matter how simple it is. I am running in a Docker container (if that matters).

TravisHC, 2021-04-20:
Hi, the same issue wth me when I launched a URDF model in gazebo:
[Err] [ModelDatabase.cc:390] Unable to parse model.config for model[http://gazebosim.org/models/bin_4_dropping_task]

Giggus, 2021-05-18:
Same here. Any solution on this?

Logan1904, 2021-05-21:
Same here. I get this error: [Err] [ModelDatabase.cc:390] Unable to parse model.config for model[http://gazebosim.org/models/bin_4_dropping_task]. This is followed by many more errors of unable to load models and parse model.config files after waiting a while. Anyone has solved this yet?

Giggus, 2021-05-21:
Temporary workaround I am using is:
1 - clone models from https://github.com/osrf/gazebo_models.git in ~/.gazebo/models 
2 - set GAZEBO_MODEL_DATABASE_URI to ""
Honestly it's just to avoid the error showing up and basically still have access to the models locally... not great but working

nrkumar93, 2021-05-28:
Same here. I am trying to launch from inside a docker container and @Giggus solution does not work for me :/

Giggus, 2021-05-28:
@nrkumar93 mine it's a workaround not a solution ;)
Btw I am working in a docker env as well. You need to do both steps at image build time.

In your Dockerfile:
```
# Loading gazebo model locally to avoid connection to http://gazebosim.org 
RUN mkdir ~/.gazebo \
 && cd ~/.gazebo \
 && git clone https://github.com/osrf/gazebo_models.git models
COPY ./setup.sh /usr/share/gazebo-<your_version>/setup.sh
COPY ./setup.sh /usr/share/gazebo/setup.sh

```
where setup.sh is a file that you create which basically overrites the one that you got after installing gazebo.
Mine for example is;
```
export GAZEBO_MASTER_URI=http://localhost:11345
export GAZEBO_MODEL_DATABASE_URI=
export GAZEBO_RESOURCE_PATH=/usr/share/gazebo-9:${GAZEBO_RESOURCE_PATH}
export GAZEBO_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/gazebo-9/plugins:${GAZEBO_PLUGIN_PATH}
export GAZEBO_MODEL_PATH=/usr/share/gazebo-9/models:${GAZEBO_MODEL_PATH}
export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
export OGRE_RESOURCE_PATH=/usr/lib/x86_64-linux-gnu/OGRE-1.9.

```

Hope it helps

nrkumar93, 2021-05-28:
Thanks @Giggus. that works

ad1tyat, 2021-06-01:
Thanks @Giggus , I was trying to run gazebo through a docker container ([this one](https://hub.docker.com/r/koenlek/ros-nvidia))
and was getting the same errors, your solution/workaround worked !

j-rivero, 2021-06-01:
Hello, thanks for the report and workarounds. I can look into the problem. Does anyone have a Dockerfile at hand to reproduce the problem? Which versions of Gazebo are you using?

Giggus, 2021-06-01:
This might be the basic docker deriving from nvidia docker env

```
FROM nvidia/cudagl:11.1.1-base-ubuntu18.04
 
# Minimal setup
RUN apt-get update \
 && apt-get install -y locales lsb-release
ARG DEBIAN_FRONTEND=noninteractive
RUN dpkg-reconfigure locales
 
# Install ROS Melodic
RUN sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
RUN apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
RUN apt-get update \
 && apt-get install -y --no-install-recommends ros-melodic-desktop-full
RUN apt-get install -y --no-install-recommends python-rosdep
RUN rosdep init \
 && rosdep fix-permissions \
 && rosdep update
RUN apt-get install -y --no-install-recommends \
	python-rosinstall\
	python-rosinstall-generator\
	python-wstool
RUN echo "source /opt/ros/melodic/setup.bash" >> ~/.bashrc

```
Running Gazebo 9

piradata, 2021-09-11:
Anyone has an solution for this? I'am still having this error too..

nm46nm, 2021-11-18:
The same problem.
On Ubuntu 16.04 and gazebo 7.
I want to make PX4 simulations. When I `make px4_sitl_default gazebo` this problem accurs.
I have copy all models in PX4(Firmware/Tools/models) to ~/.gazebo but it doesn't work.

ipez, 2021-11-21:
> The same problem. On Ubuntu 16.04 and gazebo 7. I want to make PX4 simulations. When I `make px4_sitl_default gazebo` this problem occurs. I have copy all models in PX4(Firmware/Tools/models) to ~/.gazebo but it doesn't work.

I have the same problem too. Did you solve it?

nm46nm, 2021-11-22:
> > The same problem. On Ubuntu 16.04 and gazebo 7. I want to make PX4 simulations. When I `make px4_sitl_default gazebo` this problem occurs. I have copy all models in PX4(Firmware/Tools/models) to ~/.gazebo but it doesn't work.
> 
> I have the same problem too. Did you solve it?

Not yet. Sad. You are also in ubuntu 16 and px4?

kintzhao, 2022-02-21:
git clone https://github.com/osrf/gazebo_models  
copy -rf ./gazebo_models/bin_4_dropping_task  ~/.gazebo/models/bin_4_dropping_task

**make GAZEBO_MODEL_DATABASE_URI to ""**

sudo vim   /usr/share/gazebo/setup.sh  
export GAZEBO_MODEL_DATABASE_URI="" # http://gazebosim.org/models

 cat  /usr/share/gazebo/setup.sh 

export GAZEBO_MASTER_URI=http://localhost:11345
export GAZEBO_MODEL_DATABASE_URI="" # http://gazebosim.org/models
export GAZEBO_RESOURCE_PATH=/usr/share/gazebo-9:${GAZEBO_RESOURCE_PATH}
export GAZEBO_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/gazebo-9/plugins:${GAZEBO_PLUGIN_PATH}
export GAZEBO_MODEL_PATH=/usr/share/gazebo-9/models:${GAZEBO_MODEL_PATH}
export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
export OGRE_RESOURCE_PATH=/usr/lib/x86_64-linux-gnu/OGRE-1.9.0

uhobeike, 2022-05-13:
> Hey guys, I have been trying to load a DEM in gazebo just like it is shown on the Gazebo DEM tutorial. (http://gazebosim.org/tutorials) Everytime I start Gazebo in the terminal it won't even show an [...]

Perhaps we have found a solution.

In my case, I got the following error output.

```
[Err] [ModelDatabase.cc:235] No <database> tag in the model database database.config found here[http://gazebosim.org/models/]
[Err] [ModelDatabase.cc:294] Unable to download model manifests
```

The ``http://gazebosim.org/models/`` site apparently cannot be opened properly with a 404 error.

As a solution, the following site seems to be normal at the moment.
``http://models.gazebosim.org/`

Set an environment variable to make this site do the downloading.
This idea was inspired by the code in this section.
`https://github.com/osrf/gazebo/blob/94afbb1cf9286fe21ae21eefe42ccaeaa0f0f839/gazebo/common/ModelDatabase.cc#L121

So I re-set the environment variable with the following command.
```
export GAZEBO_MODEL_DATABASE_URI="http://models.gazebosim.org/"
```

And after running roslaunch, the models were downloaded securely under `~/.gazebo/models`.

Translated with www.DeepL.com/Translator (free version)