gazebosim/gazebo-classic #2938: Gazebo Windows post-install error 0xc0000142

FirefoxMetzger, 2021-02-26:
I managed to compile Gazebo 11 from source on Windows in VS 2019 (yay 🎈).

However, after the installation, I get an error: `The application was unable to start correctly (0xc0000142)` for both `gzserver` and `gzclient`. This is true for both, the Release and the Debug build. I know that this relates to missing (or incompatible) dependencies, but I can't figure out what's wrong.

Dependency Walker shows that all the VC redistributable libraries are missing (known problem, dependency walker can't handle API sets), but shows that the Gazebo specific dependencies (sdformat, ignition-..., etc.) are present.

[Dependencies](https://github.com/lucasg/Dependencies) can handle API sets, and shows that most of the VC redistributable libraries are present. The same is true for the Gazebo-specific dependencies. What seems to be missing on my system is the library pointed to by `ext-ms-win-shell32-shellcom-l1-1-0.dll` (presumably a windows library).

My problem is that I can't find any viable information online about what that library is part of and how I can install it. I was hoping that one of you has come across a similar issue before and can give me a pointer. Any help is appreciated :)

traversaro, 2021-04-20:
Hi @FirefoxMetzger ! Sorry, I had missed this issue. 

> Dependency Walker shows that all the VC redistributable libraries are missing (known problem, dependency walker can't handle API sets), but shows that the Gazebo specific dependencies (sdformat, ignition-..., etc.) are present.

How did you installed the dependencies of Gazebo? The `0xc0000142` error could unfortunately come from deep down Gazebo's dependency stack, that has some quite non-trivial corners (FreeImage for one : ) ). Out of the methods listed in https://github.com/osrf/gazebo/issues/2901 to install Gazebo dependencies, both vcpkg (that provides both Release and Debug builds) and conda-forge (that only provides Release builds) should work fine.

> [Dependencies](https://github.com/lucasg/Dependencies) can handle API sets, and shows that most of the VC redistributable libraries are present. The same is true for the Gazebo-specific [...]

I experienced this kind of issues with `Dependencies`, and that may be simple a false positive/negative on `Dependencies` side, and not the actual source of your `0xc0000142` .

FirefoxMetzger, 2021-04-27:
In the end, I wiped the entire installation and switched to using WSL2. 

WSL2 has the limitation that rendering is CPU accelerated (llvmpipe) and uses at most 8 cores, but with a strong enough CPU that is okay for development. Simple scenes like a panda pick'n'place run at ~70-80% real-time on my machine. In addition, GPU acceleration for WSL2 is on the horizon, too, so it is just a matter of time until this becomes a non-issue. You can get it in the bleeding edge Windows build, but last I checked it was still unstable on my particular machine.

> How did you installed the dependencies of Gazebo?

I installed what was available from vcpkg via vcpkg, and - for the ignition packages not yet on vcpkg - I cloned the version matching the one from the dome (at the time) stack.

> Out of the methods listed in #2901 [...] conda-forge (that only provides Release builds) should work fine.

It does, and it does launch Gazebo. Unfortunately, it seems to suffer from a model loading issue ( https://github.com/osrf/gazebo/issues/2899#issuecomment-785682234 ). This might be fixed now though; I haven't checked since.

> I experienced this kind of issues with Dependencies, and that may be simple a false positive/negative on Dependencies side

Interesting. I didn't know that Dependencies has the same struggle. Thanks for the heads up!

traversaro, 2021-05-19:
I guess we can close this issue if all doubts have been closed. @sandyUni probably you can write in Gazebo Discourse (https://community.gazebosim.org/) for generic feedback or a separate issue for running Gazebo under WSL2? Thanks!

FirefoxMetzger, 2021-05-19:
Fair point @traversaro . I'll close it.

@sandyUni That is exciting, indeed. I need to retest if the insider build is now stable on my machine; I haven't tested it again.

I think the 30 FPS are a limitation coming from Gazebo itself because a lot of parts are tuned to run with a target RTF of 1. It's possible to set it to a higher value, but last I checked that this wasn't guaranteed to scale sensor update rates appropriately and would only change physics, which might make controllers unstable or have other unwanted side-effects.

closed 2021-05-19 by FirefoxMetzger