gazebosim/gazebo-classic #2942: Potentially incorrect gearbox simulation in Gazebo 9

vmanoj1996, 2021-03-03:
I have a gearbox joint setup between two gear links and a gear base link. Gearbox joint constrains the motion between first and second gear as expected. However when gear's base is rotated by applying torque, the gears do not turn as expected. It seems like the gearbox joint is working relative to global frame instead of gear's base.

[SDF snippet](https://gist.github.com/vmanoj1996/577a27bd16aa2ffa3e4db957bd112306)

Shared a [video for reference](https://www.youtube.com/watch?v=2AMS1-87unc)

Hoping to get more clarity on this issue. Thanks in advance!

Maxx84, 2021-04-15:
FYI, I was not able to reproduce your issue on Gazebo 11.
See [video](https://youtu.be/GXarabos-ZY).

Maxx84, 2021-04-15:
Same with 9.16.0.