gazebosim/gazebo-classic #2946: Gazebo 9/11 on macOS catatalina 10.15.7 won't load freeimage

lucarag, 2021-03-08:
Hi,

So, basically I'm trying to install gazebo on a macbook pro with macOS catatalina 10.15.7.
I successfully install it (either version 11 or 9) with
brew install gazebo11 (or gazebo9 - of course when I try to install the alternative version I remove the former first)
when I try running gazebo, however, there is this error:

$ gazebo
dyld: Library not loaded: /usr/local/opt/freeimage/lib/libfreeimage.dylib
Referenced from: /usr/local/bin/gazebo
Reason: Incompatible library version: gazebo requires version 3.0.0 or later, but libfreeimage.dylib provides version 0.0.0
Abort trap: 6

Now, I checked at the path above and the required library is actually present (screenshot below). Even try reinstalling it with brew (brew install and brew reinstall freeimage) or downloading the library and installing it manually or coping the files in that path does not fix the problem. This same files and paths can be found in Cellar. 
I quite sure the problems lies within the gazebo installation.

Is there a way to solve this?

Thank you in advance.
[image]

scpeters, 2021-03-08:
can you provide the output of `brew outdated`, `brew config`, and `brew doctor`?

lucarag, 2021-03-09:
Hi, thanks for answering. 

I'm attaching three .txt with outputs you requested. Let me know if I can provide any other information.

[brew_config.txt](https://github.com/osrf/gazebo/files/6108950/brew_config.txt)
[brew_doctor.txt](https://github.com/osrf/gazebo/files/6108952/brew_doctor.txt)
[brew_outdated.txt](https://github.com/osrf/gazebo/files/6108954/brew_outdated.txt)

lucarag, 2021-03-24:
@scpeters Hi, sorry tagging you. are these outputs helpful? do you have any idea?

Thank you in advance.

scpeters, 2021-03-26:
I noticed the following in the `brew doctor` output:

~~~
Uncommitted files:
  ?? Formula/libfreeimage-3.15.1-osx.rb
~~~

are you using the standard version of `freeimage` in homebrew-core? what does `brew info freeimage` show?

lucarag, 2021-03-30:
Sorry answering this late. This is the output:

```
$ brew info freeimage
freeimage: stable 3.18.0 (bottled), HEAD
Library for FreeImage, a dependency-free graphics library
https://sourceforge.net/projects/freeimage
/usr/local/Cellar/freeimage/3.18.0 (16 files, 16.0MB) *
  Poured from bottle on 2021-03-06 at 22:30:36
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/freeimage.rb
License: FreeImage
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 1,788 (30 days), 5,528 (90 days), 18,789 (365 days)
install-on-request: 197 (30 days), 691 (90 days), 2,461 (365 days)
build-error: 0 (30 days)
```

scpeters, 2021-04-06:
can you try `otool -L /usr/local/opt/freeimage/lib/libfreeimage.dylib` to see what version info is embedded in the `dylib`?

lucarag, 2021-04-06:
sure. here it is:

`
$ otool -L /usr/local/opt/freeimage/lib/libfreeimage.dylib
/usr/local/opt/freeimage/lib/libfreeimage.dylib:
	@rpath/libfreeimage.3.dylib (compatibility version 3.0.0, current version 3.18.0)
	@rpath/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
	@rpath/libjpeg.9.dylib (compatibility version 13.0.0, current version 13.0.0)
	@rpath/libtiff.5.dylib (compatibility version 11.0.0, current version 11.0.0)
	@rpath/libpng16.16.dylib (compatibility version 54.0.0, current version 54.0.0)
	@rpath/libopenjp2.7.dylib (compatibility version 7.0.0, current version 2.3.1)
	@rpath/libwebpmux.3.dylib (compatibility version 4.0.0, current version 4.4.0)
	@rpath/libwebp.7.dylib (compatibility version 8.0.0, current version 8.4.0)
	@rpath/libjxrglue.0.dylib (compatibility version 0.0.0, current version 1.1.0)
	@rpath/libraw.19.dylib (compatibility version 20.0.0, current version 20.2.0)
	@rpath/libIlmImf.24.dylib (compatibility version 24.0.0, current version 24.0.0)
	@rpath/libIex.24.dylib (compatibility version 24.0.0, current version 24.0.0)
	@rpath/libHalf.24.dylib (compatibility version 24.0.0, current version 24.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1197.1.1)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
`

lucarag, 2021-04-06:
sure, here it is:

```
$ otool -L /usr/local/opt/freeimage/lib/libfreeimage.dylib
/usr/local/opt/freeimage/lib/libfreeimage.dylib:
	@rpath/libfreeimage.3.dylib (compatibility version 3.0.0, current version 3.18.0)
	@rpath/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
	@rpath/libjpeg.9.dylib (compatibility version 13.0.0, current version 13.0.0)
	@rpath/libtiff.5.dylib (compatibility version 11.0.0, current version 11.0.0)
	@rpath/libpng16.16.dylib (compatibility version 54.0.0, current version 54.0.0)
	@rpath/libopenjp2.7.dylib (compatibility version 7.0.0, current version 2.3.1)
	@rpath/libwebpmux.3.dylib (compatibility version 4.0.0, current version 4.4.0)
	@rpath/libwebp.7.dylib (compatibility version 8.0.0, current version 8.4.0)
	@rpath/libjxrglue.0.dylib (compatibility version 0.0.0, current version 1.1.0)
	@rpath/libraw.19.dylib (compatibility version 20.0.0, current version 20.2.0)
	@rpath/libIlmImf.24.dylib (compatibility version 24.0.0, current version 24.0.0)
	@rpath/libIex.24.dylib (compatibility version 24.0.0, current version 24.0.0)
	@rpath/libHalf.24.dylib (compatibility version 24.0.0, current version 24.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1197.1.1)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
```

scpeters, 2021-04-06:
that is a little different from what I see on my machine running 10.15:

~~~
$ otool -L /usr/local/opt/freeimage/lib/libfreeimage.dylib 
/usr/local/opt/freeimage/lib/libfreeimage.dylib:
	/usr/local/opt/freeimage/lib/libfreeimage.dylib (compatibility version 3.0.0, current version 3.18.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 902.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)
~~~

can you try removing and reinstalling `freeimage` from the homebrew bottle?

lucarag, 2021-04-07:
ok so I removed and reinstalled `freeimage` (deleting the bottle cached in homebrew downloads first).
The output is now this:

```
$ otool -L /usr/local/opt/freeimage/lib/libfreeimage.dylib
/usr/local/opt/freeimage/lib/libfreeimage.dylib:
	/usr/local/opt/freeimage/lib/libfreeimage.dylib (compatibility version 3.0.0, current version 3.18.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 902.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)
```

but when I try running gazebo I still get the same error:

```
$ gazebo
dyld: Library not loaded: /usr/local/opt/freeimage/lib/libfreeimage.dylib
  Referenced from: /usr/local/bin/gazebo
  Reason: Incompatible library version: gazebo requires version 3.0.0 or later, but libfreeimage.dylib provides version 0.0.0
Abort trap: 6
```

lucarag, 2021-05-05:
any suggestions?