gazebosim/gazebo-classic #2959: Gazebo doesn't load meshlab scaled models and doesn't show any error

Darkproduct, 2021-04-05:
## TLDR:
Scaling any `.obj` mesh with meshlab will make it without any error output even with `--verbose` not loadable by gazebo.

## Steps to reproduce:
I`m using this simple stone model: https://sketchfab.com/3d-models/rock-stone-02-0455747c7ac848269830ed0dca33726c

After downloading, extracting and some renaming my folder structure looks like this:
```
.
├── meshes
│   ├── stone_01.mtl
│   └── stone_01.obj
├── model.config
├── model.sdf
└── textures
    ├── stone_01.jpg
    ├── stone_01_normal.jpg
    └── stone_01_occlusion.jpg
```
`model.sdf`:
```
<?xml version="1.0" ?>
<sdf version="1.6">
  <model name='Stone_01'>
    <pose frame=''>0 0 0.2 0 -0 0</pose>
    <static>true</static>
    <allow_auto_disable>true</allow_auto_disable>
    <link name='Stone_01_link'>
      <collision name='collision'>
        <pose>0 0 0 0 -0 0</pose>
        <geometry>
          <box>
            <!-- TODO: Better collision box. Maybe an ultra low poly rock. -->
            <size>1.516384 0.974134 0.495299</size>
          </box>
        </geometry>
        <max_contacts>10</max_contacts>
      </collision>
      <visual name='visual'>
        <geometry>
          <mesh>
            <uri>model://stone_01/meshes/stone_01.obj</uri>
            <scale>1 1 1</scale>
          </mesh>
        </geometry>
        <material>
          <!--<shader type="normal_map_object_space">
          <shader type="normal_map_tangent_space">
            <normal_map>model://stone_01/textures/stone_01_normal.jpg</normal_map>
          </shader>-->
          <pbr>
            <metal>
              <albedo_map>model://stone_01/textures/stone_01.jpg</albedo_map>
              <ambient_occlusion_map>model://stone_01/textures/stone_01_occlusion.jpg</ambient_occlusion_map>
              <normal_map type="tangent">model://stone_01/textures/stone_01_normal.jpg</normal_map>
            </metal>
          </pbr>
        </material>
      </visual>
    </link>
  </model>
</sdf>
```

The stone will be loadable with its texture, but I'm not sure if the ambient_occlusion_map and the normal_map is working properly. But still. There is a model with texture. 
[image]
Output from starting with `rosrun gazebo_ros gazebo --verbose` and adding the model from the `insert` tab.
```
Gazebo multi-robot simulator, version 11.4.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[ INFO] [1617629722.264775503]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1617629722.267743523]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.178.112
[Msg] Loading world file [/usr/share/gazebo-11/worlds/empty.world]
[ INFO] [1617629722.603234672, 0.010000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1617629722.628483938, 0.034000000]: Physics dynamic reconfigure ready.
```

Scaling with meshlab. Load the `.obj` file into meshlab. Use `Filters ⇒ Normals, Curvatures and Orientation ⇒ Transform: Scale, Normalize`. Change the values for each axis (I used 0.5 on each) and click apply. Now go to `File ⇒ Export Mesh…` and hit ok.
Trying to load the `.obj` file now will result in nothing, but you'll be able to select it which shows an empty bounding box:
[image]
Output from starting with `rosrun gazebo_ros gazebo --verbose` and adding the model from the `insert` tab.
```
Gazebo multi-robot simulator, version 11.4.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[ INFO] [1617632110.687954177]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1617632110.689942239]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.178.112
[Msg] Loading world file [/usr/share/gazebo-11/worlds/empty.world]
[ INFO] [1617632111.023999576]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1617632111.055724540, 0.014000000]: Physics dynamic reconfigure ready.
```

---
I'm using Ubuntu 20.04 LTS with ROS Noetic, Gazebo 11.4.0.

newcanopies, 2021-04-24:
while it could be a bug in Gazebo 11.4, 
you might find some clues if you compare the original to the meshlab-scaled .OBJ mesh in a text editor.
meshlab could be introducing an encoding change to the obj file

mintar, 2023-06-08:
I've debugged this and found a solution. You have to edit your .mtl file.

The Blender-generated file (working in Gazebo) has this line:

    d 1.0

The Meshlab-generated file (broken in Gazebo) has this line:

    Tr 1.000000

This defines transparency, see [Wikipedia](https://en.wikipedia.org/wiki/Wavefront_.obj_file#Basic_materials):

> Materials can be transparent. This is referred to as being *dissolved*. Unlike real transparency, the result does not depend upon the thickness of the object. A value of 1.0 for "d" is the default and means fully opaque, as does a value of 0.0 for "Tr". Dissolve works on all illumination models. 

    # some implementations use 'd'
    d 0.9
    # others use 'Tr' (inverted: Tr = 1 - d)
    Tr 0.1

In other words, the Meshlab export specifies full transparency, which Gazebo correctly renders as "invisible".

**Solution:** Remove the "Tr 1.000000" line (or better yet, copy the complete lines from the original Blender file) and the error is fixed.